FormerLurker/ArcWelderPlugin #88: Test part

5axes, 2020-12-05:
On the Benchy boat model, is there a model that could be used as a test part to validate conversions in G02/G03 interpolation?

FormerLurker, 2020-12-05:
Good question...  I don't have anything that is perfect for testing.  I have a couple gcode files I use for testing.  I have a spiral, a benchy sliced with Archimedean infill and have cut out the first layer, as well as a few files that caused issues in the past.

5axes, 2020-12-05:
This one can be used for different test but as I have no special issue in G1 with the sliced Gcode it's caoont be considered as a rela test part.
[image]
Hereafter the OpenSCAD file :
[SpiraleCercle.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5647255/SpiraleCercle.zip)

FormerLurker, 2020-12-05:
Nice!  I have a small circle scad that I just remembered (just got up.. head fog) that i will post here.

FormerLurker, 2020-12-05:
[arc_test.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5647353/arc_test.zip)

[image]

FormerLurker, 2021-02-06:
Hey, going to close this.  The file you sent is VERY useful, thank you so much!

5axes, 2021-02-06:
No need to Jeep it open

closed 2021-02-06 by 5axes