FormerLurker/ArcWelderPlugin #89: Octoprint Software Update Error

andrewluebke, 2020-12-06:
In Octoprint 1.5.1 (and 1.5.0) on Raspberry Pi I get the following error:

[image]

in the log I see:

2020-12-06 10:05:17,762 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1496, in _get_current_version
    target, check, online=online, credentials=credentials
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 305, in get_latest
    target
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None

I've tried reinstalling Arc Welder with the same result.  This is sort of a cosmetic problem for me since the plugin still works fine and I could reinstall by hand if I want to update but it would be nice to have Octoprint tell me when a new version is available.

FormerLurker, 2020-12-06:
This is fixed in the latesy devel branch, which I am still working on.  It is pretty close to a devel/rc.  I will post a link here so you can test the update fix.  Thanks for your report!

andrewluebke, 2020-12-06:
Thanks, I just saw the other issue report in closed.  I'll try it when you have a link for me.

holocronology, 2020-12-10:
Still doing this in the 1.5.2 hotfix update.

FormerLurker, 2022-05-17:
Sigh...  I was finally packaging up this version for a deploy, but this is a showstopper.  Will get on it.  Thanks for the report since this is probably the most difficult thing to test for.