FormerLurker/ArcWelderPlugin #90: TypeError: remove_file() takes exactly 3 arguments (2 given)

bvrielink, 2020-12-10:
When processing a gcode file it fails and gives the following error in the log file:

```
2020-12-10 12:23:09,776 - arc_welder.preprocessor - ERROR - An unhandled exception occurred while preprocessing the gcode file.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_arc_welder/preprocessor.py", line 229, in run
    self._process(task)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_arc_welder/preprocessor.py", line 326, in _process
    task, results
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_arc_welder/__init__.py", line 947, in preprocessing_success
    task, results, additional_metadata
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_arc_welder/__init__.py", line 749, in save_preprocessed_file
    self._file_manager.remove_file(target_path)
TypeError: remove_file() takes exactly 3 arguments (2 given)
 
```

This is with version 1.0.1rc1.dev2+u.2a95fab. Here are the most important settings regarding this error:

[image]

bvrielink, 2020-12-10:
Workaround for now is to disable overwrite source file:

[image]

Then the processing is successful.

FormerLurker, 2020-12-10:
I could have sworn this was working.  I must have somehow removed the LOCAL flag...  Will fix and push asap, thanks for the report!

FormerLurker, 2020-12-10:
This is strange.  Take a look at the docs for that command:

[image]

and the corresponding definition:

[image]

What version of OctoPrint are you using?

bvrielink, 2020-12-10:
OctoPrint 1.5.1
Python 2.7.16
OctoPi 0.17.0

FormerLurker, 2020-12-11:
Ok, can you install via the plugin manager by clicking 'Get More' and pasting this link into the 'from url...' box:  https://github.com/FormerLurker/ArcWelderPlugin/archive/8754b74bb5255eb52e67bd8e04e95404bba4b61d.zip

Let me know if that helps.

bvrielink, 2020-12-11:
Just started another print for that other issue. Will try it after it's finished. Thanks.

bvrielink, 2020-12-11:
Tested the zip file and the problem seems to be solved. What did you change?

FormerLurker, 2020-12-11:
I added the missing parameter :)  The docs I saw were referring to the storage interface, and the file manager docs were missing.  Not sure how or why, but I found the commit where I removed the storage location parameter, and I guess I never tested with overwrite after that (it's inconvenient when testing the same file over and over to have to reupload it).

FormerLurker, 2020-12-11:
It's just one of those goofs that happen with a one man/woman show, lol.

closed 2021-02-06 by FormerLurker