FormerLurker/ArcWelderPlugin #91: ArcWelderPlugin prints the wrong file

bvrielink, 2020-12-11:
I uploaded a file from Cura to OctoPrint. It's being processed and starts to print afterwards. These are the settings for the plugin:

[image]

You can see it has created a new file:

[image]

But it prints the uploaded file and not the new file created by the plugin:

[image]

FormerLurker, 2020-12-11:
Do you have the 'print after upload' option enabled within the cura plugin by chance?  If so, disable that

FormerLurker, 2020-12-11:
I will do that.  There is a cura arc welder plugin coming out soon, fyi.

closed 2020-12-11 by FormerLurker