FormerLurker/ArcWelderPlugin #92: Arc welder incorrectly adjusting z axis

PixelOrange, 2020-12-11:
Printer: Ender 5+ Stock
Marlin: 1.2.1
Cura: 4.8.0
Arc welder version: 1.0.0+u.bb71e8f
G2/G3 test successful

Description: Some arc welder prints cause the first few layers to adjust the z, causing the nozzle to touch the bed. The non-arc welded gcode works fine. This doesn't happen on every print. Some prints work just fine with the arc welded version. It seems to happen on larger prints more often than smaller ones. I assume this is due to the z axis not adjusting as much on prints under 10 cm size.

I would like the ability to tell arc welder to not touch z at all. I only need the x/y adjusted.

FormerLurker, 2020-12-11:
Hmm.. Your version of arc welder does not even use Z coordinates at all.  Can you send me an example of an unprocessed gcode file that failed?

PixelOrange, 2020-12-11:
I foolishly deleted the aw.gcode that was having an issue and I'm mid print so I can't run it to see if the newly generated file does the same thing or not. I've attached both the aw.gcode and the original .gcode.

[crossbar.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5681153/crossbar.zip)

FormerLurker, 2020-12-13:
@PixelOrange, I have plotted your gcode on several viewers, and I cannot figure out what is happening.  The code looks fine to me.  If you see this on any print in the future, please snag the before and after gcode and send it over.  It sounds like only the first layer is affected, so it should be pretty easy to see what's going on.

So, the newest unreleased version of arcwelder CAN adjust Z, so I'm really interested in seeing how this is happening on your version that cannot output z coordinates.  The issue you are seeing is still possible if arc welder somehow swallowed a gcode and didn't output it.

PixelOrange, 2020-12-18:
Attached are the .stl, .gcode, and .aw.gcode files. I started the .aw.gcode and noticed the following:

* Initial wipe was fine.
* Half of skirt was fine, the other half adjusted z. Unlike when I reported this issue, the z moved once, printed half the skirt, then moved back for the other half. When I initially reported this, z gradually moved as it was printing. As far as I know, I do not have mesh turned on.
* First layer was fine but it appeared to do a z-hop at the end of each wall. z-hop is not enabled. The actual layer did not change, just hop and then immediately print again.

I have not attempted to print the non-aw.gcode. I have not updated any of my software, still the same as when I reported this issue.

[tree-holder.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5718036/tree-holder.zip)

PixelOrange, 2020-12-24:
I printed another object and noticed the z adjust. Stopped it and ran the non-aw.gcode. I noticed that when it went to wipe, it slowly adjusted z as the X/Y was moving until it got to where it was going to wipe. It did the full wipe, and then moved back to center to do the skirt and again adjusted z as it was moving to where it wanted to start the skirt. Here's what I suspect is happening:

X/Y/Z are all being adjusted at the same time in my initial gcode. I don't know how or why this is enabled. ArcWelder is seeing that as a curve and attempting to do its thing.

If I can find out what is causing the z-adjust as it gets into position, I believe I can fix this issue.

FormerLurker, 2020-12-25:
I will take a look asap.  Been stuck at work for the last couple of weeks.  Tha ks for the update, it may prove useful

PixelOrange, 2021-01-04:
I found the Z line adjust in the start g-code in Cura. Arc appears to not mess with that at all. I don't see another Z adjust until layer 1. I'm considering removing the Z line adjust in the starting code just to see if the behavior continues but at this point I'm not sure it's arc that's causing the issue. I just don't understand why it only happens with arc prints.

FormerLurker, 2021-01-04:
I think I finally understand what is going on.  I will be reviewing this soon, probably tomorrow.

FormerLurker, 2021-01-04:
@PixelOrange,

> As far as I know, I do not have mesh turned on.
This line suggests you have both mesh enabled (loaded from memory) and that you have z fade turned on:

```M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height```

I'm guessing you have a BL Touch installed?  If so, remove that line and try it again.  You should be able to just delete it from the arcwelded gcode and run it again.

If that fails I suggest you try the original non-welded gcode, and try printing the first 3mm of the print.  You should notice the Z axis will stop adjusting after 2mm (due to the M420 Z2 parameter).

FYI, I looked at each and every Z coordinate adjustment in both files, and there is nothing wrong there.  Hopefully removing that command will solve your issue.  If you want to use mesh bed leveling, I would recommend manually sending a mesh leveling command, then store the results in eeprom like so:

```
G29 ; Initiate mesh bed leveling
M500; Save the results of the mesh bed leveling to EEPROM so it's there when you restart the printer
```

Alternatively, if you want to re-level your bed every time you start a print (takes a bit longer, but will be more accurate since your bed could shift slightly for various reasons), just replace this line in your start gcode:

```
G28 ; Home
```
with this 
```
G29 ; Initiate mesh bed leveling
G28 X0 Y0 ; Home X and Y
```

Regardless of which method you use above, you should now be able to use the M420 command to initiate Z fade.  Just note that Z WILL adjust as your extruder moves about due to the mesh.  If it's moving a lot, you may need to manually adjust your bed and run the mesh leveling again.

I hope that helps!

PixelOrange, 2021-01-04:
I do have a BLtouch. It came with the printer. I'll give that a shot and let you know. I don't want to use mesh leveling as my bed is fairly level. None of my prints run through the mesh points at the beginning of the print, just the initial check to find the bed height.

Thank you!

PixelOrange, 2021-01-12:
I was finally able to test this. The saved mesh was the issue. I wasn't aware there was a saved mesh so when I releveled I never ran a new one. Problem solved, thank you!

closed 2021-01-12 by PixelOrange