FormerLurker/ArcWelderPlugin #93: Bad G2/G3 command from ArcWelder

Misterke, 2020-12-11:
I just had a print fail a couple of hours in, due to a bad G2 command according to Marlin.  My octoprint log contained the following:

```
| Send: N26938 G1 X97.767 Y171.872 E961.9324*100
| Recv: ok
| Send: N26939 G2 X97.766 Y171.871 I-0 J-0 E961.93251*87
| Recv: Error:G2/G3 bad parameters
2020-12-11 16:48:40,861 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Error: G2/G3 bad parameters"
2020-12-11 16:48:40,895 - octoprint.util.comm - INFO - Force-sending M112 to the printer
2020-12-11 16:48:40,989 - octoprint.plugins.filamentmanager - INFO - Filament used: 9307.263119999983 mm (tool0)
```

I haven't been in-depth with actual g-code myself, but I guess Marlin complains about both I and J being zero.  I can provide the complete original g-code file if needed, but I would assume the input and resulting sequence could be sufficient, so ...

In the original file I find:
```
G0 F12000 X122.327 Y172.052
G1 F3000 X122.207 Y172.052 E960.3972
G1 X122.207 Y171.873 E960.4082
G1 X115.626 Y171.878 E960.8127
G1 X111.274 Y171.871 E961.0802
G1 X104.715 Y171.877 E961.48334
G1 X98.367 Y171.874 E961.87352
G1 X98.367 Y172.052 E961.88446
G1 X97.767 Y172.052 E961.92134
G1 X97.767 Y171.872 E961.9324
G1 X97.767 Y171.871 E961.93247
G1 X97.766 Y171.871 E961.93251
G1 X92.382 Y171.878 E962.26344
G1 X89.001 Y171.871 E962.47125
G1 X33.112 Y171.872 E965.90645
```
which after welding seems to become:
```
G0 F12000 X122.327 Y172.052
G1 F3000 X122.207 Y172.052 E960.3972
G1 X122.207 Y171.873 E960.4082
G2 X98.367 Y171.873 I-10.716 J28697.69 E961.87351
G1 X98.367 Y172.052 E961.88446
G1 X97.767 Y172.052 E961.92134
G1 X97.767 Y171.872 E961.9324
G2 X97.766 Y171.871 I-0 J-0 E961.93251
G3 X89.001 Y171.871 I-4.382 J-1300.232 E962.47125
G1 X33.112 Y171.872 E965.90645
```

FormerLurker, 2020-12-11:
What version are you using?  That is crazy.  I don't think this could happen in the latest build, but I will definitely test.

Misterke, 2020-12-11:
I am running a devel version https://github.com/FormerLurker/ArcWelderPlugin/commit/2a95fab3a7797b3dad3664bece27f08137d0f46c mainly because of earlier mention that this one was supporting the "thumbnail copying" and I confirm that it is ...

FormerLurker, 2020-12-11:
Yikes.. Ok.  Looks like I have some research to do.

Misterke, 2020-12-11:
I wonder whether it is a rounding error.  The curve ending with I and J zero seems to be a very tiny deviation from the previous G1 coordinates.  So probably I / J would not really be zero but very close to zero - possibly too close?

FormerLurker, 2020-12-11:
can you post the entire file?  The welding process is somewhat dependent on the preceding code.  Also, a snapshot of the Preprocessor Settings are also necessary, so I can make sure the exact same arcs get generated.

FormerLurker, 2020-12-11:
Use gist.github.com if you need a place to post the file, then leave a link in here.  Dropbox or whatever else you may have access to is also fine.

Misterke, 2020-12-11:
Original file can be found here: https://gist.github.com/Misterke/500b220a39dc10c898caaa45cdb0adc8

I ran it through ArcWelder a second time and also that run delivered the erroneous G2 command ...

FormerLurker, 2020-12-11:
So, more mystery.  Here is what I'm getting running with your settings:

G0 F12000 X122.327 Y172.052
G1 F3000 X122.207 Y172.052 E960.3972
G1 X122.207 Y171.873 E960.4082
G2 X98.367 Y171.873 I-10.716 J28697.69 E961.87351
G1 X98.367 Y172.052 E961.88446
G1 X97.767 Y172.052 E961.92134
G1 X97.767 Y171.872 E961.9324
G1 X97.767 Y171.871 E961.93247
G3 X33.112 Y171.872 I-32.684 J-23072.3 E965.90645

However, I am debugging via windows, so it's totally possible slight differences in FP precision or optimization could lead to the result you are having.  I will run this on my py here soon, but am first going to take a look at the algorithm to see if I can detect a potential 0 for both I and J.  Also, I did find several instances where either I or J was 0, but not both, but that is sort of expected.  Hopefully I will have something for you soon.

jbbandos, 2020-12-12:
I have the same exact issue while printing "soot sprites", a couple of hours into the print it fails with "Error: G2/G3 bad parameters". I can post both original and aw gcodes if you want, the original is 32.5MB though.

hoefs, 2020-12-12:
Had the same issue here with 1.0.1rc1.dev2+u.2a95fab 
My Mandalorian print stopped after 4 hours with this message in Octoprint = Bad G2/G3 command

Been using ArcWelder for al my latest prints. Never had an issue with 1.0.0...

FormerLurker, 2020-12-12:
Yes, please post the original gcode please.  This will help me to verify the solution.

hoefs, 2020-12-12:
> Yes, please post the original gcode please. This will help me to verify the solution.

Is there a way to recuperate it, because I already removed it from OctoPrint...

FormerLurker, 2020-12-12:
@hoefs, i guess you could try reslicing with the same settings (is this cura 4.7 by chance).  If This appears in the file, then it would be useful for me:

I0 J0

hoefs, 2020-12-12:
> @hoefs, i guess you could try reslicing with the same settings (is this cura 4.7 by chance). If This appears in the file, then it would be useful for me:
> 
> I0 J0

I use cure 4.8.0. I will try it later this day.

Misterke, 2020-12-12:
> @hoefs, i guess you could try reslicing with the same settings (is this cura 4.7 by chance). If This appears in the file, then it would be useful for me:
> 
> I0 J0

Note that in my case the output was actually `I-0 J-0`, so I guess any combination with or without `-` for either of the parameters would result in a failure report from Marlin.

FormerLurker, 2020-12-12:
@Misterke, yes, you are correct.  So that means there are four things to search for @hoefs:

```I0 J0```
```I-0 J0```
```I0 J-0```
```I-0 J-0```

Any of these will trigger the issue.  While I'm at it I will just exclude I and J if they = 0 within the resolution of the gcode.

FormerLurker, 2020-12-12:
Well, I've figured out how this error crept in.  After the 1.0 release I removed some shoddy math that was compensating for an issue I thought was precision related.  Turns out i was rounding a number too aggressively later in the code.  The old 'fix' was enforcing a minimum radius.  After removing that Arc Welder was able to convert many more small radius arcs.  Unfortunately the fix was so good that arcs of a radius below the precision of the gcode were being produced.

Another complication is that I am dynamically updating the maximum precision of the gcode output based on the input (with 3 digits being the minimum).  So if gcode appears with 5 decimals of precision, the output precision also gets updated.

I believe I've handled this by adding a 'tolerance' that is based on the precision of X Y and Z.  So if the precision is 3, the tolerance would be 0.001.  Any values less than the tolerance will be considered 0.  If both abs(I) and abs(J) are less than tolerance, the arc will be aborted.  I could take rounding into consideration and handle values that are 1/2 the tolerance, since those will get rounded up..  I will think about that, but I don't think it would make much difference.

FormerLurker, 2020-12-12:
Also @Misterke, I confirmed your error when running this code on the PI.  It's unfortunate that I'm not yet able to create this through my debugger, but at least I can test it on the pi.

hoefs, 2020-12-13:
@FormerLurker this i found in my code:

`G1 X118.391 Y96.18 E3452.55179
G2 X118.618 Y96.582 I4.624 J-2.35 E3452.56238
G2 X118.62 Y96.584 **I0 J0** E3452.56249
G3 X123.04 Y94.368 I9.735 J13.909 E3452.72753
G3 X129.334 Y95.489 I2.106 J6.4 E3452.94879
G1 X129.683 Y95.205 E3452.96378`

You ca download the file here:
https://hoefs.no-ip.com:5001/d/s/592383625106865197/T3ZfkNZFTkQ3JkWWNyFpW_sV-46QlFNV-bbxATyCSOAg_

Do you have a new beta/rc with a fix? (running 1.0.1rc1.dev2+u.2a95fab now)

Misterke, 2020-12-13:
I've switched to the latest commit from the devel branch and 'rewelded' the problematic file and ... no more zero I and J.  Comparing the generated gcode:

```
After correction:                                     Before correction:
G0 F12000 X122.327 Y172.052                           G0 F12000 X122.327 Y172.052
G1 F3000 X122.207 Y172.052 E960.3972                  G1 F3000 X122.207 Y172.052 E960.3972
G1 X122.207 Y171.873 E960.4082                        G1 X122.207 Y171.873 E960.4082
G2 X98.367 Y171.873 I-10.716 J28697.69 E961.87351     G2 X98.367 Y171.873 I-10.716 J28697.69 E961.87351
G1 X98.367 Y172.052 E961.88446                        G1 X98.367 Y172.052 E961.88446
G1 X97.767 Y172.052 E961.92134                        G1 X97.767 Y172.052 E961.92134
G1 X97.767 Y171.872 E961.9324                         G1 X97.767 Y171.872 E961.9324
G1 X97.767 Y171.871 E961.93247                        G2 X97.766 Y171.871 I-0 J-0 E961.93251
G3 X33.112 Y171.872 I-32.684 J-23072.3 E965.90645     G3 X89.001 Y171.871 I-4.382 J-1300.232 E962.47125
```
I've already printed this model without welding, so won't be passing it through Marlin, but at least the gcode seems fine...
Thanks for the quick fixing!

hoefs, 2020-12-13:
> I've switched to the latest commit from the devel branch and 'rewelded' the problematic file and ... no more zero I and J. Comparing the generated gcode:
>
> ```
> After [...]

How can I install this latest commit via OctoPrint?

hoefs, 2020-12-13:
@Misterke Found it... (after I had another failed print 2 hours in) I downloaded the devel zip file and installed in OctoPrint.
I am running 1.0.1rc1.dev2+u.8779945 now. Hope this one finishes the print...

FormerLurker, 2020-12-13:
@Misterke  and @hoefs,

You guys jumped the gun a bit :)  I have put in prob 24 hrs into debugging this issue as well as another caused by obscenely high precision in some inputs that was overflowing the marlin gcode buffer slot length.  To resolve the other issue I had to add a few parameters:  Default XYZ precision, Default E precision and Allow Dynamic Precision.  That was unfortunate, but necessary.  Most people will never need to touch the defaults of 3 for XYZIJ and 5 for E (so G2 Xx.xxx Yx.xxx Zx.xxx Ix.xxx Jx.xxx Ex.xxxxx Fx) with dynamic precision off (added for smoothieware and other firmware that supports more than 3 digits).

Regarding your issue, there were a few cases that were still not fixed according to my testing, though I think the version you installed is mostly OK (I run some extreme tests).  

I also learned that many viewers seem to have a problem if either I or J is omitted, so I forced output for all coordinates, which has little impact on the final gcode size anyway.  Better to NOT have viewers showing bad renderings and lose a KB or two per 50MB of gcode in compression.

Also, I had been working on some performance improvements involving some math enhancements (removing pow, pre-calculating tolerances, etc), and using a new modified double to string function (based on fpconv, modified to produce fixed precision rounded output).  The result has been to approximately cut the processing time in half.  I decided to merge this in since you two seem pretty capable.

Try reinstalling from this link and let me know what you think:  https://github.com/FormerLurker/ArcWelderPlugin/archive/8964fbe92181139a4a191cb4137b6644f7985bef.zip

I'm continuing to test.  Keep in mind this is a devel version!  I would love to hear if you notice any increases in performance (and issues too, but I won't love those as much, lol)!  Thanks!

hoefs, 2020-12-14:
@FormerLurker 

Successfully uninstalled Arc-Welder-1.0.1rc1.dev2+u.8779945
Successfully installed Arc-Welder-1.0.1rc1.dev2+u.8964fbe

Misterke, 2020-12-14:
@hoefs, I just do an install plugin from URL https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip to get the latest devel version.

I've again done that just now (which gives me v1.0.1rc1.dev2+u.8964fbe and re-welded the problem file again and can confirm that also with this commit there are no arcs with both I and J 0.  But as said, I currently don't need to print that model, so won't be passing it through Marlin ...

hoefs, 2020-12-16:
@FormerLurker I only printed some litho's and an Among Us character, no issues with this build

hoefs, 2020-12-18:
@FormerLurker I am 10 prints later, no issues seen. I can't really say if there is an increase in performance...

FormerLurker, 2020-12-18:
@hoefs, there should be some reduction for time to process the gcode, but not print time.  Is that what you mean?  Thanks for reporting success!  Fyi, there is still one issue I am working on.  Will post the issue number later.

hoefs, 2020-12-18:
@FormerLurker yes you are right, it is faster. If I need to try a new version, please let ne know...

btw this is impressive:

[image]

closed 2021-02-06 by FormerLurker