FormerLurker/ArcWelderPlugin #94: Arcwelder causing a Checksum mismatch error

eblieb, 2020-12-12:
Recv: Resend: 25
Recv: ok
Send: N25 G3 X68.871999999999999 Y87.040999999999996 I9.860755608737562 J0.060818139804325 E8.01619*95
Recv: Error:checksum mismatch, Last Line: 24

Generated a retraction tuning Gcode from this site - https://teachingtechyt.github.io/calibration.html#retraction and ran it through arc welder and it causes a checksum mismatch error. 

Using this firmware - https://github.com/smith3d/Marlin/releases/ - E3V2-2.0.x.14 Slow 5x5
Ender 3 V2

Disabled Arc Welder, Reuploaded the gcode and it works fine without arc welder processing it

FormerLurker, 2020-12-12:
This came from arcwelder gcode:

N25 G3 X68.871999999999999 Y87.040999999999996 I9.860755608737562 J0.060818139804325 E8.01619*95

???  That kind of resolution should never be produced.  Can you send me the source gcode as well as the converted gcode?  Also, what version of the arc welder plugin are you using (it's reported on the tab)?

eblieb, 2020-12-12:
https://1drv.ms/u/s!AuWXqoE3r4cjgfd21ZzvYENMt9mVXg?e=jFeP8e - This is the gcode before arcwelder manipulated it. I dont have a copy of the gcode after arc welder did its thing because I disabled arc welder and reuploaded the file over the invalid checksum file

Arc Welder v1.0.1rc1.dev2+u.8754b74

FormerLurker, 2020-12-12:
Can you re-weld it with your current settings and post that please?

eblieb, 2020-12-12:
https://1drv.ms/u/s!AuWXqoE3r4cjgfd3o6AHm571bDIB7Q?e=vQNq4s - Here is the welded gcode

FormerLurker, 2020-12-12:
Ok, so this appears to be a limitation of the checksum algorithm.  Arc welder dynamically sets the maximum resolution of the gcode based on the maximum resolution found in the source file.  I'm seeing lines like this in your source:

G1 X72.096 Y86.17699999999999 E0.1445
G1 X72.809 Y86.13900000000001 E0.1753

So based on those lines, arc welder will produce output with 15 decimal places for X,Y,Z,I and J, like so:

G2 X126.519000000000005 Y92.487999999999999 I-0.44600000599793 J6.24712499550958 E12.69099

Note that E precision is handled separately.

It looks like the resulting gcode has more characters than is allowed by your firmware.

Not sure what to do about this.  I've never seen code with that much precision.  It may work for G1/G0 since there are fewer characters there due to the lack of I and J parameters.

I will think about this, and may make the dynamic resolution optional, though I think the retraction tool (I'm assuming that is where the before gcode came from) should not produce that much precision.

eblieb, 2020-12-12:
Not sure what language plugins are written in, but could always set the maximum amount of decimal places if they exceed a set maximum resolution of the firmware. If firmware is set to 10 decimal places, if resolution exceeds 10 then use 
cout << setprecision(10) << value; for C++
limited_Number = round(Number, 10) for Python

Would something like that work?

FormerLurker, 2020-12-12:
It's in c++, and that is one option to make it work.  I think I'll add three new parameters (yuck, lol):

Allow Dynamic Resolution (false by default)
Default XYZ Precision (3 by default)
Default E Precision (5 by default)

FormerLurker, 2020-12-13:
@eblieb,

I think I have this licked.  I have put in prob 24 hrs into debugging this issue as well as another caused by rounding arcs of a very small offsets that resulted in values of 0 for both I and J.  That was a similarly severe issue to the one you saw, since it killed the print once it hit these lines.  

To resolve your issue I had to add a few parameters:  Default XYZ precision, Default E precision and Allow Dynamic Precision.  The defaults will work fine in your case (3 for xyz, 5 for e, dynamic precision off).

I also learned that many viewers seem to have a problem if either I or J is omitted, so I forced output for all coordinates, which has little impact on the final gcode size anyway.  Better to NOT have viewers showing bad renderings and lose a KB or two per 50MB of gcode in compression.

Also, I had been working on some performance improvements involving some math enhancements (removing pow, pre-calculating tolerances, etc), and using a new modified double to string function (based on fpconv, modified to produce fixed precision rounded output).  The result has been to approximately cut the processing time in half.  I decided to merge this in since you seem to be a pretty capable tester.

Try reinstalling from this link and let me know what you think:  https://github.com/FormerLurker/ArcWelderPlugin/archive/8964fbe92181139a4a191cb4137b6644f7985bef.zip

I'm continuing to test.  Keep in mind this is a devel version!  I would love to hear if you notice any increases in performance (and issues too, but I won't love those as much, lol)!  Thanks!

FormerLurker, 2021-02-06:
Have you been able to try this out to verify the fix?  I'm pretty sure it's solved, but always like to have verification from OP, thanks!

eblieb, 2021-02-06:
Ya I haven't seen any issues and been using It in my prints since you fixed
it

FormerLurker, 2021-02-06:
Awesome, thank you for getting back so quickly!  Closed!

closed 2021-02-06 by FormerLurker