FormerLurker/ArcWelderPlugin #96: Firmware compatibility for Marlin bugfix-2.0.x

hoefs, 2020-12-13:
HI,

Is there a way to make this firmware "officially" compatible with ArcWelder?

[image]

I use firmware for my Ender 3 V2 from https://marlin.crc.id.au/

My firmware includes arc support:

[image]

FormerLurker, 2020-12-13:
I would recommend suggesting to the firmware provider that they include an actual version number, and not a general 'bugfix-2.0.x` version.  The problem here is that there's no way to distinguish 2.0.0 (not fully supported/recommended, though it works) from 2.0.6 (fully supported, works great).  I have also seen versions of this branch that simply do not work at all, and have been modified so that the automatic detection I do doesn't work properly.

I'm really not sure what to do about this except maybe switching 'unsupported' to 'unknown'...

mriscoc, 2020-12-27:
Hi @FormerLurker, I made a firmware based upon current Marlin bugfix. I know that it supports Arc Welder plugin, but I only has been got to be recognized by the plugin when I raised the firmware version. What are your recommendations for the correct version label for bugfix firmware releases? I want to follow your guidelines.
Firmware: https://github.com/mriscoc/Marlin_Ender3v2/releases/tag/2.0.7.3-bugfix
[image]

FormerLurker, 2020-12-27:
Just set it to the forked marlin version, that should work.  Fyi, I am going to change the bugfix branch from 'not supported' to 'unknown' by default, but if you set the marlin version as I mentioned, it will display as supported