FormerLurker/ArcWelderPlugin #97: Feature Request: Confirmation dialog for manual processing

KenLucke, 2020-12-15:
Can you add a [possibly optional via prefs) confirmation dialog for a manual process of a file?  I've accidentally clicked the "Welds Arcs" button too many times now :).

Also, an option to retain the final result (the green one) dialog on screen until dismissed.  It goes away too fast for me, I end up having to stare at the screen until it appears and screenshot it quickly, if I want to look at any of the details.

FormerLurker, 2020-12-15:
What version are you using right now?

> Can you add a [possibly optional via prefs) confirmation dialog for a manual process of a file? I've accidentally clicked the "Welds Arcs" button too many times now :).

Will think about this.  Shouldn't be too difficult.

> Also, an option to retain the final result (the green one) dialog on screen until dismissed. It goes away too fast for me, I end up having to stare at the screen until it appears and screenshot it quickly, if I want to look at any of the details.

I think this has been solved in the latest build.  I include all of the processing info on the tab in the statistics, including conversion times, and much more.

KenLucke, 2020-12-15:
My current version is Arc-Welder (1.0.0+u.bb71e8f) 

I had not seen any updates since that.

KenLucke, 2020-12-15:
> I think this has been solved in the latest build. I include all of the processing info on the tab in the statistics, including conversion times, and much more.

Damn, sorry, completely forgot that info was there.

FormerLurker, 2020-12-15:
> I had not seen any updates since that.
Yeah, I've not released any, but I've made 52 different commits since that release.  Wanna try the devel build?  It should be reasonably stable, and there are only a couple known issues (mostly related to overwriting the source file).  If you want to check it out, install from this url via the plugin manager:  https://github.com/FormerLurker/ArcWelderPlugin/archive/8964fbe92181139a4a191cb4137b6644f7985bef.zip

I could use more testers :)

FormerLurker, 2020-12-15:
Here is a snap of the new tab:

[image]

KenLucke, 2020-12-15:
> Yeah, I've not released any, but I've made 52 different commits since that release. Wanna try the devel build? It should be reasonably stable, and there are only a couple known issues (mostly [...]

I'll give it a shot.  I stopped using AW a while back because, although it was great on arcs, it messed up straight areas.  I recently have been using it manually on some heavily curved models, where there are virtually no straight stretches.

KenLucke, 2020-12-15:
With (left) and Without (Right) AW.  Same file (thus same settings), merely Arc Welded.  Would probably be even better, but was printing at 100mm/s

[image]

FormerLurker, 2020-12-15:
Awesome results, thanks for posting!

Not sure how straight segments could be messed up, but it may be worth looking into.  Arcwelder doesn't mess with those (at least if it is working properly).  Could be firmware related maybe?  I'm curious what the firmware checker says about it.

KenLucke, 2020-12-15:
Oh-oh.  Now I'm getting this when attempting to manually convert:
[image]
 Log is empty.

From config.yaml:
``` 
    arc_welder:
        _config_version: 2
        allow_dynamic_precision: true
        feature_settings:
            delete_source: always
            file_processing: manual-only
        firmware_compensation_enabled: true
```

Also tried turning off dynamic precision, also restarting Octoprint.

KenLucke, 2020-12-15:
> Not sure how straight segments could be messed up, but it may be worth looking into. Arcwelder doesn't mess with those (at least if it is working properly). Could be firmware related maybe? I'm curious what the firmware checker says about it.

[image]

It's Nic Wilson's upgraded Marlin 2.0.7.5, and it worked fine before I just upgraded to your development version :( .

KenLucke, 2020-12-15:
Backed off to your v1.0.0 and it's working again

FormerLurker, 2020-12-15:
And, thats why things get tested.  Send me that gcide if you would please.  Will look at the marlin fw report if you run an m115 command and send me the results from the terminal.

KenLucke, 2020-12-15:
> And, thats why things get tested

Yup, been alpha/beta testing stuff since the early 80s.  I understand the need to test, and the obscurity of some bugs.

>  Send me that gcode if you would please.

https://www.dropbox.com/s/newm17fxjews27b/Meteor%20Light%20Holder%20v2%20%5BPETG%5D%20%5Bskirt%5D%20%7C%20Infill%20%5B20%25%2Ccubic%5D%20%7C%20Noz%20%5B0.6%5D%20%7C%20LH%20%5B0.32%5D%20%7C%20LW-%5B0.6%5D%20%7C%20Temps%20%5B255%C2%B0%7C%2080%C2%B0%5D%20%7C%20Speed%20%5B100.0%5D%20%7C%20Support%20%5Bnormal%20%2859%C2%B0%29%5D%20%7C%20Coast-%5BFalse%5D.gcode.zip

>  Will look at the marlin fw report if you run an m115 command and send me the results from the terminal.

```
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.7.5_NIC NICW VEADBB (Nov 23 2020 10:16:54) SOURCE_CODE_URL:https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0 PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR10 V2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:1
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok P15 B7
Send: M155 S2
Recv: ok P15 B7
Send: M876 P1
Recv: ok P15 B7
```

Going into printing mode now for several hours, so can't test anything on any of the machines that would require the machine in idle mode.

FormerLurker, 2020-12-15:
The dropbox link you sent doesn't seem to work.  That filename is pretty crazy (I can't say I've ever seen anything like it), so perhaps that is an issue.  To post it maybe rename it to something less..  complicated?

FormerLurker, 2020-12-15:
It could be browser related too.  The pipe symbol isn't allowed in any windows filename (and can trigger script errors in linux).

KenLucke, 2020-12-15:
> The dropbox link you sent doesn't seem to work. That filename is pretty crazy (I can't say I've ever seen anything like it), so perhaps that is an issue. To post it maybe rename it to something less.. complicated?

Sorry, it's probably all the spaces and brackets.

https://www.dropbox.com/s/newm17fxjews27b/Meteor%20Light%20Holder.zip

FormerLurker, 2020-12-15:
OK, when you have a second and aren't printing, I pushed a new version to devel that should log the error you saw.  You can install it with this URL:  https://github.com/FormerLurker/ArcWelderPlugin/archive/2e839051de8d81da0a7ae3374028c0c416fc852f.zip

My intrusion prevention blocked dropbox after seeing that file request, lol!  I'll unblock it and download it ASAP to try myself, but I'd still like to see what is logged after you try it out.  I did test the filename (minus the characters that the OS doesn't allow), and that worked on the devel build, so it must be something else.

KenLucke, 2020-12-15:
will do.  Earliest printer free is in ~ 6 hours.