FormerLurker/ArcWelderPlugin #98: Feature Request: Start print after conversion

Futureless671, 2020-12-17:
I print to Octoprint directly from Cura, so I would love to see an option to automatically load and print the file after it's converted. Right now, the original file is uploaded, the print is started, and then it's converted, resulting in the original file being printed instead of the converted one. I'm no programmer, but I think the easiest way to implement this is to disable the "start print" option in Cura, and then just have an arc welder option to start the print after the file is converted.

o-nix, 2020-12-17:
@Futureless671 already implemented in `dev` version https://github.com/FormerLurker/ArcWelderPlugin/issues/46

Futureless671, 2020-12-17:
Ah, my bad. Thanks!

closed 2020-12-17 by Futureless671