FormerLurker/ArcWelderPlugin #100: ArcWelderConsole large gaps in walls

DaveZA, 2020-12-18:
using ArcWelderConsole.exe on win8.1pro.

Been working flawlessly, except for some large cylindrical models ( like large pots ).
The walls ( inner and outer ) have gaps about 10mm wide, 4 per circumference.
I am guessing that ArcWelder is splitting the circular shape in to 4 sections, and the gap appears between sections.
Diameter of the models is between 150 and 210 mm.
Infill in the areas is still being printed.
Seems that it only appears where the wall of the pot is vertical.

[image]

FormerLurker, 2020-12-18:
Interesting.  Can you post the before and after gcode please?  I am working on another issue, and i think they may be related.

DaveZA, 2020-12-18:
Hi
Attached zip of the 2 files, and also batch file output from ArcWelderConsole.exe
Not isolated to a single model. I am doing a range of these and happening on different models.

Also, please can you point me to where I find the ArcWelderConsole.exe file ? Can't see it anywhere on github.

Regards
Dave

[ArcOutput.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5713922/ArcOutput.txt)
[Rounded Large.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5713914/Rounded.Large.zip)

DaveZA, 2020-12-18:
ok. I found the exe in the ArcWelderLib

FormerLurker, 2020-12-18:
Awesome.  There could be a few issues with the new binaries.  Keep your eyes peeled for an update to the artifacts.

DaveZA, 2020-12-18:
that would be the file Windows-MSVC.zip in the ArcWelderLib ?

FormerLurker, 2020-12-18:
Yes.  I need to clean up the artifacts, just haven't had time.  No need for an msvc and mingw version, and mingw is broken (I would need to add static linking, but why bother when I can just remove it, lol)

DaveZA, 2020-12-18:
> Yes. I need to clean up the artifacts, just haven't had time. No need for an msvc and mingw version, and mingw is broken (I would need to add static linking, but why bother when I can just remove it, lol)

to a non-technical person, that is as 'clear as mud' :)

FormerLurker, 2020-12-25:
@DaveZA, sorry it's taken me so long to work my way back to your issue (work is always nuts right before the new year).  I tried looking at your zip file, and it was flagged by google as containing malware.  I ran my anti-virus on it, and it said the same thing, so I'm not going to open it.  I recommend you scan your PC.

Can you post the before and after on gist.github.com instead and drop a link in here?  That should be just fine.

DaveZA, 2020-12-25:
> @DaveZA, sorry it's taken me so long to work my way back to your issue (work is always nuts right before the new year). I tried looking at your zip file, and it was flagged by google as containing [...]

very odd. 
I also downloaded the zip file and got the same warnings. 
I have already deleted the files from my PC a few days back.
This will have to wait until I have a new model that presents the same problems, although I have already decided to upgrade the printer to an SKR 1.4 Turbo 32 bit board in the first week of the new year, so there's also a good chance that this will no longer be an is
sue

FormerLurker, 2020-12-25:
Ok, that is fine!  Let me know how it goes!

FormerLurker, 2020-12-28:
@DaveZA, I'm going to close this issue for now, but if you continue to have problems with the new board I will reopen it.  Thanks!

closed 2020-12-28 by FormerLurker