FormerLurker/ArcWelderPlugin #101: AW Displays invalid warning

KenLucke, 2020-12-23:
Both welded and unwelded files display this warning if you hover over the icon in the file list while a print job is in progress (note: it also shows an "uunwelded" icon even for those files that HAVE been welded (the one in the image has been):

[image]

HOWEVER, despite the warning, it still DOES allow me to process the file, even while printing (I've noticed no impact, but then I'm running a Pi 4b with little else, so it's still got lots of computational power to spare). [the ability to do this should be a preference in the settings, anyway]

https://user-images.githubusercontent.com/57936294/103039443-ca980900-4525-11eb-85b1-1a3abe2bf987.mov
```
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:84.0) Gecko/20100101 Firefox/84.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3927818240
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin 2.0.7.5_NIC NICW VEADBB (Nov 23 2020
```

FormerLurker, 2020-12-25:
What version of arc welder is this?  Check the tab and paste in the exact version found on the bottom left.  I haven't been able to reproduce this in the devel branch code, but things have changed quite a bit over the last month or so.

Also, since OctoPrint and all associated plugins and processors run on a single thread, it can and will cause issues if you process while printing depending on the gcode you are printing, so i don't plan to allow users to process while printing.  This would cause me far more headaches than it would be worth for the minority of users processing on something powerful enough to support this (A pi 4 can definitely have issues too, as I have verified in my testing).  Instead, the newest version will queue up files and will stop any processing if a print starts.  The queue will start back up as soon as printing is finished.  See this shot:

[image]

I think this is a good solution for everyone, since it won't affect any print on any hardware, and it also allows files to be chosen for welding while a print is going.   What do you think?

KenLucke, 2020-12-25:
Nothing on version lower left.  However, the OTHER left (lower RIGHT [grin]) does have 
[image]

FormerLurker, 2020-12-25:
Ok, try this version and see if this fixes your issue (install from plugin manager->get more->from url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/e7e7d46e86d4f5c6ba77022f5e26a0511942260b.zip``

Warning:  if things go wrong you may need to uninstall with the clean option and downgrade.  Be sure to review the settings because a lot has changed, and the settings migration is probably the least tested part.