FormerLurker/ArcWelderPlugin #105: Arc Welder not working properly

Athanaetos, 2020-12-27:
I have an Ender 3 Pro, with an upgrade SKR Mini 2.0 board. Arc welder used to work on my old board, but now it isn't working anymore.

It seems to disregard any skirt or brim and goes straight for the build and proceeds to under extrude and make something that doesn't resemble the actual file.

Non arc welded Gcode works perfectly. I wonder what the issue is, I would love to get Arc Welder working again as it was a saving grace on my old motherboard. 

Thanks!

Athanaetos, 2020-12-27:
I might have done an oopsie and not enabled Arc in firmware. However I was certain its enabled by default. Reinstalling Marlin and trying a test print. Will update this post if its my dumb.

Athanaetos, 2020-12-27:
Was my oopsie. Left a 4 hour print on and it made a mess and so I made this post. Remember to check with the M115 to see if Arc is on!

closed 2020-12-27 by Athanaetos

FormerLurker, 2020-12-28:
Hey @Athanaetos, if you would be willing, perhaps you could try the latest devel branch (I'm pushing out an rc/devel release as we speak with the same code)?  I've added a firmware checker that will hopefully mostly solve this issue, which is the most common one I see.  If you are willing, you can try it by going to OctoPrint Settings->Plugin Manger->Get More->From Url.. and pasting this in the box:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/a0f41e1ae75443805278b643ed9162bfcae17394.zip```

I'd be interested in seeing what the firmware checker reports as well as the results of sending an M115 to your printer.  Thanks!

Athanaetos, 2020-12-29:
I will give it a shot and update here.