FormerLurker/ArcWelderPlugin #106: Feedback for 1.0.1rc1.dev2

FormerLurker, 2020-12-28:
If you have any feedback for the [1.0.1rc1.dev2](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1rc1.dev2) release, please leave that here. Let me know what you like, what you love, what you hate, or any ideas about the future of Arc Welder. If you believe you have found a bug, feel free to mention it here. I may ask you to create a new issue.

Thank you in advance for your contribution!

CRCinAU, 2020-12-28:
Trying to install the new build, but things are failing with:
```
    GNU C++11 (Debian 8.3.0-6) version 8.3.0 (aarch64-linux-gnu)
    	compiled by GNU C version 8.3.0, GMP version 6.1.2, MPFR version 4.0.2, MPC version 1.1.0, isl version isl-0.20-GMP
    
    GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
    Compiler executable checksum: 0deabfc33a693f28b24a1b5260e3137f
    In file included from octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp:27:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/fpconv.h:55:28: error: conflicting declaration ‘typedef long long unsigned int uint64_t’
     typedef unsigned long long uint64_t;
                                ^~~~~~~~
    In file included from /usr/include/stdint.h:37,
                     from /usr/lib/gcc/aarch64-linux-gnu/8/include/stdint.h:9,
                     from /usr/include/c++/8/cstdint:41,
                     from /usr/include/c++/8/bits/char_traits.h:503,
                     from /usr/include/c++/8/string:40,
                     from octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.h:24,
                     from octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp:22:
    /usr/include/aarch64-linux-gnu/bits/stdint-uintn.h:27:20: note: previous declaration as ‘typedef __uint64_t uint64_t’
     typedef __uint64_t uint64_t;
                        ^~~~~~~~
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp: In static member function ‘static bool utilities::get_file_path(const string&, std::__cxx11::string&)’:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp:251:28: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<std::__cxx11::basic_string<char> >::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
      for (int index = 0; index < file_parts.size() - 1; index++)
                          ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp: In static member function ‘static std::__cxx11::string utilities::dtos(double, unsigned char)’:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp:298:8: warning: unused variable ‘p’ [-Wunused-variable]
      char* p = buffer;
            ^
    In file included from octoprint_arc_welder/data/lib/c/gcode_processor_lib/utilities.cpp:27:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/fpconv.h: At global scope:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/fpconv.h:149:11: warning: ‘Fp find_cachedpow10(int, int*)’ defined but not used [-Wunused-function]
     static Fp find_cachedpow10(int exp, int* k)
               ^~~~~~~~~~~~~~~~
    error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    ----------------------------------------
```

FormerLurker, 2020-12-28:
Let me see if I can deal with that.  I had to remove an include to get the dumb and old python 2.7 compiler for windows working...

FormerLurker, 2020-12-28:
@CRCinAU, try installing from this URL:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip```

This may be the shortest lived devel/rc in history, lol!

CRCinAU, 2020-12-28:
```
Building wheels for collected packages: Arc-Welder
  Building wheel for Arc-Welder (setup.py) ... done
  Created wheel for Arc-Welder: filename=Arc_Welder-1.0.1rc1.dev2+u.2dedd1b-cp37-cp37m-linux_aarch64.whl size=1575364 sha256=99185727e2e2cab254e578abd00f3642f382b8ed9994b6fa9b0e220f65c5f9a9
  Stored in directory: /tmp/pip-ephem-wheel-cache-4p7jp23c/wheels/bf/41/e0/cd84677fe6b976b644ba40cf7bf0c0bd2dfd581ab35ecd68ed
Successfully built Arc-Welder
Installing collected packages: Arc-Welder
Successfully installed Arc-Welder-1.0.1rc1.dev2+u.2dedd1b
```

Restarted OctoPrint and there it is:
[image]

FormerLurker, 2020-12-28:
Cool.  I will be so glad when python 2 is totally dead, lol!

CRCinAU, 2020-12-28:
I hear you man..... I hear you....

If there's one thing I do well is find problems :P

o-nix, 2021-01-14:
1. Couldn't install via the initial link, only by commit ID, Python is 3.7.3
2. The option says `Show Show Current Run Configuration`
3. I wiped out the settings before installing the new version, enabled the "Overwrite" option, started a print and it does not show the stats for the processed file; maybe because I didn't restart yet?

FormerLurker, 2021-01-14:
@o-nix, not sure why you had install issues, but the statistics not showing up was a race condition.  I will post another release soon.

o-nix, 2021-01-14:
@FormerLurker maybe because I didn't delete the previous RC version first?

FormerLurker, 2021-01-14:
@o-nix , it shouldn't have mattered unless maybe you upgraded your python version?  there is an issue with the upgrade script ppl have been using that could cause some problems.

o-nix, 2021-01-14:
@FormerLurker no, I fully wiped OctoPrint's installation ~2 months ago, almost everything is out of the box. I didn't save the logs, sorry, but everything went good through `pip3` and it was saying something about "installation in error state" at the very end. Then I deleted rc1, restarted, and successfully installed `2dedd1b9dfa5da4e75e968107473a8113369e709.zip`

pmos69, 2021-01-24:
Running 1.0.1rc1.dev2+u.cb5e627 on an FLSun QQ-S Pro (repetier) and after quite a few prints it keeps on working great, with default settings.

Source File:FQQS_Ponteira_joana.gcode
Target File:FQQS_Ponteira_joana.aw.gcode
Arcs:Created 4.7K from 82.9K extrusions
Target/Source Extrusions:5.8K ⁄ 84.0K = 93.1% reduction
Compression:5.8∶1 (82.7%)
File Size:2.6 M - 2.1 M = 455.2 K
Line Count:95.8K - 82.9K = 12.9K
Processing Time:3 seconds
Segment Length Statistics:
   Min          Max     Source  Target  Change
----------------------------------------------
  0.000mm to   0.002mm       7       6  -14.3%
  0.002mm to   0.005mm       5       0 -100.0%
  0.005mm to   0.010mm       4       0 -100.0%
  0.010mm to   0.050mm   11004      36  -99.7%
  0.050mm to   0.100mm    1388      12  -99.1%
  0.100mm to   0.500mm   25716     293  -98.9%
  0.500mm to   1.000mm   45506    3111  -93.2%
  1.000mm to   5.000mm     270    1725  538.9%
  5.000mm to  10.000mm      68      84   23.5%
 10.000mm to  20.000mm       5      38  660.0%
 20.000mm to  50.000mm       1      95 9400.0%
 50.000mm to 100.000mm       0     407     INF
          >= 100.000mm       0       0    0.0%
----------------------------------------------
Total distance source:.............39192.747mm
Total distance target:.............39197.306mm
   Total count source:...................83974
   Total count target:....................5807
 Total percent change:..................-93.1%

FormerLurker, 2021-01-24:
Wow, that is some major compression!  I'm prepping dev3 now, which will hopefully be the last dev rc.

FormerLurker, 2021-02-06:
OK, closed!  Thanks everyone who commented!

closed 2021-02-06 by FormerLurker