FormerLurker/ArcWelderPlugin #107: Creality CR10S Pro V2 and TinyMachines3d.com

stonefox1958, 2020-12-28:
Can I use it with :
Creality CR10S Pro V2 and 
TinyMachines3d firmware

FormerLurker, 2020-12-28:
Hmm. Not sure.  You could try the latest RC (see the release feedback issue for details) and see what the firmware checker says.

stonefox1958, 2020-12-29:
What does RC means ?
There can I find firmware checker ?

FormerLurker, 2020-12-29:
It stands for 'Release Candidate'.  If no major bugs are detected, it will be promoted from a Development RC to a Maintenance RC, then to a live version.

You can install from the plugin manager by clicking 'Get More', then by pasting this URL into the 'From URL..." box:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip```

FormerLurker, 2020-12-29:
FYI, the firmware checker will show up on the ArcWelder tab, and should look something like this:

[image]

FormerLurker, 2021-02-06:
Have you had a chance to try this out yet?  Just checking in.

darsys, 2021-05-14:
I am running this on a Creality CR10S Pro V2 but I am using Desus Firmware [here](https://github.com/Desuuuu/Marlin/tree/bugfix).  Works great but be sure to enable arc support in Marlin 2.0.