FormerLurker/ArcWelderPlugin #108: Temperatur problem with big arcs

realloser, 2020-12-29:
Hi, has anyone also faced this problem... I'm printing a part which has big arcs around d:110mm. 
[image]

While printing this parts, the temperature of the hotend rises a lot and I hit the max temperature specified in the firmware and aborts the print.
[image]

My assumption is that the when printing outer perimeter, wich is printed slower and for big arcs can take a lot of time to execute this gcode command, the firmware is not able to correct the hotend temperature. This leads to spikes which can hit the threshold.

Is this something one can fix within marlin? If not I would suggest to introduce a max arc length parameter to segment an existing arc allowing the firmware to better monitor the hotend.

ArcWelder version: (Arc Welder v1.0.0+u.bb71e8f)[https://github.com/FormerLurker/ArcWelderPlugin/commit/bb71e8fe20cc65f0ab28c0c13685fde39f730320]
Marlin: 1.2.1 (😱 very old I know)

FormerLurker, 2021-01-03:
I have not seen this before!  I recommend trying to upgrade marlin, because arc welder does not alter any commands dealing with temperature.  Let me know how it goes, and I'll keep my eyes peeled for any similar reports.

realloser, 2021-01-21:
Never change a running system. I managed to kill my drivers but finally I have upgraded to Marlin 2.x. I didn‘t do the exact same print, but my test print looked promising. I hope this fixed it.

FormerLurker, 2021-01-21:
@realloser, hey, that's great!  I wanted to mention that I DID see [some new code dealing with temperature handling for arcs](https://github.com/MarlinFirmware/Marlin/blob/5ee1087959f88dc60386ff3caa21e75d9e20b128/Marlin/src/gcode/motion/G2_G3.cpp#L174-L178) in Marlin.  Probably related.

FormerLurker, 2021-02-06:
Hey, going to close this out.  I've gotten verification from several users that some versions of marlin are affected by this issue, so it is upstream.  Thank you for posting!

closed 2021-02-06 by FormerLurker