FormerLurker/ArcWelderPlugin #109: ArcWelder failed to load on PC without VC 2019 Runtime installed

unwiredben, 2020-12-30:
I installed the plugin from the Ultimaker Cura Marketplace, but after a restart, I couldn't find any ArcWelder-related items in the user interface.  After looking through the Cura log, I found that the problem was that ArcWelder.exe wouldn't load because I didn't have the Visual Studio 2019 runtime files installed on my PC.

I don't know if there's a way to have that prerequisite shown during install or a way to build the executable with no dependencies, but I wanted to post so other people who have the same issue could find a solution.

I ended up downloading it from https://visualstudio.microsoft.com/downloads/ and everything worked after that.

FormerLurker, 2020-12-30:
Thanks!  Will see what can be done.

FormerLurker, 2021-01-04:
Hey, I just released some binaries for windows that are statically linked to the required dlls.  If you wouldn't mind trying them out (might have to uninstall the C++ runtime) [by downloading the latest build here](https://github.com/FormerLurker/ArcWelderLib/suites/1759533024/artifacts/33499772), that would be awesome.

FormerLurker, 2021-02-06:
Were you able to give this a try?  I'm just checking in to see if I can close this issue.  It's actually a bit difficult for me to test this myself, since all my PCs are used for development, and all have the C++ runtime.  Thinking about spinning up a VM, but that does take time.

Anyway, let me know when you have some time, thanks!