FormerLurker/ArcWelderPlugin #110: [Feature Request] Move file after processing

pietchaki, 2020-12-30:
I like to keep he original gcode, but also don't like the mess it becomes in the main folder.
So, what I do is move all original gcodes to a "originals" folder, and keep only the processed files in the root.
I'd like if ArcWelder could do that moving automatically for me, to folders I specify in the configuration.

FormerLurker, 2021-01-03:
Any reason you are wanting to keep both the welded and original gcode?  Just curious.

FYI, I've been having a lot of issues deleting/moving files, and don't think I can implement this until I've worked those problems out.

pietchaki, 2021-01-08:
It's something I always did, but now that you ask why... I can't find a good reason... same goes for keeping the gcodes on octoprint for a printer that I keep changing and upgrading...

Thanks for the answer!

schwos, 2021-01-31:
+1 for this feature, I would prefer to have the original untouched file moved to a folder like /originals and the post processed file remain in the root. Then I allow another plugin move files within root to a archive folder after X days. 

What issues are you having when deleting/moving files within the code? Typically has to do with permissions but if the root folder is owned by the running owner of the process shouldn't be an issue.

FormerLurker, 2021-01-31:
@schwos, been thinking about this, and would it not be better to have a second plugin that can organize files based on various criteria, like file name, upload date, metadata, and use some kind of 'if this then that' style interface to trigger some kind of action?  Regex and wild cards could be supported.

It wouldn't be too terribly difficult to add this to arcwelder, I'm just thinking about getting the biggest bang for your buck in terms of re-use, and to provide something that might be useful to non-arc welder users.

> What issues are you having when deleting/moving files within the code? Typically has to do with permissions but if the root folder is owned by the running owner of the process shouldn't be an issue.

It's not a permission issue, it's 'file is in use' messages, and I get them randomly.  Much more often this happens in windows, but I've had the problem on the pi as well.  I have implemented a 'retry' with a delay, and that has made it more reliable, but it still fails, especially when trying to overwrite or delete the original.  I suspect other plugins or analyzers are doing things.  Arc welder processes gcode with c++ code, so it's WAY faster than similar python code, so this is likely a race condition of some kind.  I've been thinking about adding some worker thread to continue to attempt delete/rename operations, but I fear that if the delay is too long, users may be confused (why isn't it deleting, or worse yet, why did it just delete that file).  Anyway, it's an ongoing battle.

schwos, 2021-01-31:
@FormerLurker, that isn't a bad idea I am using the DeleteAfterPlugin which deletes older gCode files based upon age from the root folder. It leaves the files I put into a Archive or Original folder alone. So your proposal is a good approach. Use the If a filename with a prefix or postfix appears (say with .aw.) then find the original with no .aw. and move to original. The only thing is when I think of it is you are already opening and modify the file why not just trigger that plugin via an API to execute the move based upon what every rules are in place. 

Are you closing the handle you use when you open the file to read? I do development but new to 3D Printing and Octoprint and the plug-in architecture it provides, though seams pretty straight forward.  I have been doing a lot of project with ESP 8266 platform and wanted to print custom cases so guess what picked up another hobby.  you should be able to reopen the file, read the file and make a copy of the file in a new location, but the final step of deleting the original file; if another process is opening could delay that routine. Though I like your approach of a more advanced plugin to do file management; I saw another file management plugin but not aware of what it can do. As with anything I have loaded to many plugins which has made my Octoprint deployment turn to a "lazy dog" so I am making selective choices of what is necessary and what isn't. I made a donation for your efforts on this plugin as I compared my standard gcode output against yours on print time and quality and so far; knock on wood; it has been very impressive outcome. 

If I get through a couple of my current projects and nothing has been done with a separate plugin I may take that on as a challenge.. Thank you for the reply.

ImagineerNL, 2022-03-04:
Wanted to add a feature request, but it was already requested, so my findings here.

> @schwos, been thinking about this, and would it not be better to have a second plugin that can organize files based on various criteria, like file name, upload date, metadata, and use some kind of [...]

While this suggestion is a good route, i have noticed that when currently moving files to another folder makes the ArcWelder plugin run a new conversion on the moved filed in the new folder.
(i am using Draggable files plugin from https://github.com/SanderRonde/Octoprint-Draggable-Files )

Secondly, defining the filtering would be difficult, since i would want to keep the .aw.gcode files and move away all other files.
If i were to filter i would need to be able to filter on 'NOT "aw.gcode" '

The reason i want to keep the original is your own disclaimer stating "Arc Welder is still in development and may contain errors that could affect your prints."; so i would always like to keep a backup.

trebory6, 2023-05-04:
Hey guys, sorry to bump this, but I too would like this feature or at least a suggestion for an automatic file/folder organizing plugin. 

Now that I think of it, I think it might be a better idea for an auto file organizer plugin. That way it looks at the filename and organizes them into folders automatically by filename or certain attributes(like if it's been printed or not, successful or not, etc).

Outside of a plugin like that, I do think that in addition it'd be a good idea to have a "Move original file to folder" option within this app, as it wouldn't require a user to download an entirely new plugin. Kind of seems like a no brainer. 

Does anyone know of a plugin like this, or if not what the status of this feature request is?