FormerLurker/ArcWelderPlugin #111: Some arcs randomly printed as full circles.

Snargaloo, 2020-12-31:
I printed a model welded with an older version of ArcWelder in PETG for tolerance testing.  When I tried printing this model again in carbon filled PA using a devel branch of ArcWelder, some of the arcs sporadically printed as full circles.  Here is a picture of the PETG part (Orange) next to the PA-CF part:

[image]

Here are some close up pictures of the issue:

[image]
[image]
[image]
[image]
[image]
[image]

I am using version Arc Welder v1.0.1rc1.dev2+u.2e83905.  I sliced these models with Cura.  I build Cura from source, and this model was sliced from a build from 4.8 Origin/Master commit 9efe5dd5e29f36732148731c389c8bb280ab2402.

This is the original model gcode:
[HEAD_FRAME_eSun_ePA-CF_FR77651-.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5757918/HEAD_FRAME_eSun_ePA-CF_FR77651-.zip)

This is the ArcWelded gcode:
[HEAD_FRAME_eSun_ePA-CF_FR77651-.aw.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5757923/HEAD_FRAME_eSun_ePA-CF_FR77651-.aw.zip)

This is the ArcWelded gcode for the previous ArcWelder version (PETG):
[CE3_HEAD_FRAME.aw.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5757924/CE3_HEAD_FRAME.aw.zip)

Since this model was printed solid, I decided to trim the extra arc bits and use it.  Please let me know if I can provide any other information.

FormerLurker, 2020-12-31:
Hi there!  I think this is a duplicate issue that has been solved in the [latest devel rc](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1rc1.dev2).  See the release notes for details, and install with this link (there was a compiler issue with the previous version):  

```https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip```

I ran your gcode through the new version, and those bad arcs were eliminated.  If you want, send me your processed gcode once you install the new version and I'll run it through my analyzer to verify it's gone for you too.

TLDR:  The path tolerance percent was not being converted from a percent (say 5%) to a decimal (0.05).  This caused issues obviously, and the bug was likely introduced during a refactor.  However, not only did I fix that problem, but I added an algorithm that can detect these issues no matter what the path tolerance percent is (I used 10000% for testing).  It does slow the arc generation down somewhat, but it is the only foolproof way I've come up with for verifying the arcs without setting some kind of arbitrary limits (like the path tolerance percent).

Let me know what you think!

Snargaloo, 2021-01-01:
I have now installed Arc Welder v1.0.1rc1.dev2+u.a0f41e1.  I have not modified any other software versions and attach the original sliced gcode:
[HEAD_FRAME_eSun_ePA-CF_FR77651-(1).zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5759193/HEAD_FRAME_eSun_ePA-CF_FR77651-.1.zip)

and the welded gcode:
[HEAD_FRAME_eSun_ePA-CF_FR77651-.aw(1).zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5759195/HEAD_FRAME_eSun_ePA-CF_FR77651-.aw.1.zip)

Please process these through your analyzer.  I am about to start some prints with the new version and will report back.  Thank you for the swift response.

FormerLurker, 2021-01-01:
It looks good now!  Here is what the welded code looks like in Simplify 3D:
[image]

You can also install the *PrettyGCode* OctoPrint plugin to view welded files now, as long as you have the latest update.  Here is what it looks like in there:

[image]

I assisted updating that plugin to render arcs, and it uses a variant of Marlin's arc interpolation algorithm, so it should be quite accurate.

Thanks, and let me know if the print works out!

Snargaloo, 2021-01-01:
Is the version of Pretty GCode that you mention on the regular versioning branch?  Or is it a development version?

Snargaloo, 2021-01-01:
I just checked with the version of Pretty GCode that I have installed.  I loaded the welded gcode file that has the full circles and it renders the file just as it printed with full circles!  Thank you for the heads up!

FormerLurker, 2021-01-03:
@Snargaloo, have you tried the new arcwelder that I linked to (installed in plugin manager->get more->from url... with this link ```https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip```)?

Snargaloo, 2021-01-20:
Everything seems to be working well again, thank you!

closed 2021-01-20 by Snargaloo

FormerLurker, 2021-01-20:
woot!  Thanks for following back up, and enjoy Arc Welder!