FormerLurker/ArcWelderPlugin #112: OctoPrint Error File Not Found

PitbullRaven, 2021-01-04:
Today I noticed that when I print from Cura to OctoPrint, the files were uploaded to OP, then analised as expected, however the resulting .aw file was not being loaded after analysis and therefore the original uploaded file was being printed instead. Changed AW to delete the original in hopes this would force OP to load the calculated .aw file. However, all I get is an error from CURA. After sending to print, it uploaded, I can see AW run the analysis on the OP interface, then back in CURA I get an "OctoPrint Error File not found on 'local':whateverfilename.gcode" and as such, it won't move forward.

I know this is because OP expected to print the original uploaded file. How do I get AW to discard the original and load up the AW optimized file and then print?

FormerLurker, 2021-01-04:
The solution is to use the arc welder plugin for cura.  See if that works for you

PitbullRaven, 2021-01-04:
> The solution is to use the arc welder plugin for cura. See if that works for you

Was unaware there was a CURA plugin. Installed and will give it a go. Sorry for the stupid question. Please go ahead and kill this issue.

FormerLurker, 2021-01-04:
Lol, no prob.  It is new, and there has been a lot of discussion about this.  I hope you enjoy the plugin!

PitbullRaven, 2021-01-04:
Is the plugin just added to CURA and it'll do its thing automagically when I send a print to OctoPrint, or do I have to run it somehow before and save the file somewhere to then upload it?

FormerLurker, 2021-01-04:
Automagic :)  The best kind

FormerLurker, 2021-02-06:
So, did the cura plugin solve your problem?  Just checking in.

PitbullRaven, 2021-02-07:
Hi.

I think so... I can see the arc welder section in CURA, though I have no
idea how to check if it is working or not. I just go by faith.

I have disabled arc welder in Octoprint. Since then I no longer get error
messages.

I also changed some things in my setups, though, so hard to say what did
it. In any case I am pretty sure the plug-in resolved it.

FormerLurker, 2021-02-07:
Awesome, thanks @PitbullRaven!  Enjoy the software!  

Closed :)

closed 2021-02-07 by FormerLurker