FormerLurker/ArcWelderPlugin #114: Excude regions?

JasonLooseArrow, 2021-01-08:
Disclaimer: I have not yet upgraded to Marlin 2.0.6.
But, Arc Welder works very well when tuned to my prints. However, it distorts smaller details and I'd like to find a way to exclude some specific areas. Is there a way to do this, or do I need to upgrade my firmware? TIA!

FormerLurker, 2021-01-08:
So, the issue is 100% related to your firmware version.  However, I have solved your issue in the development version with a new feature, appropriately called 'Firmware Compensation'.  If you would like to try it out, you can install the development version via the plugin manager by clicking 'Get More' and pasting this url into the 'From Url...' box:  ```https://github.com/FormerLurker/ArcWelderPlugin/commit/a0f41e1ae75443805278b643ed9162bfcae17394```

Read [this issue](https://github.com/FormerLurker/ArcWelderLib/issues/18) which explains the new feature, what it does, and what problem it solves.  That issue is in the 'ArcWelderLib' repository, so it is targeting the .exe and .lib files, not the plugin, but I think you'll get the idea :)

It must be enabled first in the *ArcWelder* settings.  Here are the recommended defaults (notice the blue help question marks, click those for more info):

[image]

I should also mention that the UI has changed pretty dramatically, I believe it is for the better.  Additionally, the core arc generation algorithm has improved in accuracy, compression level, safety (almost no chance of bad arc generation, which was rare to begin with), and possibly speed.  It's actually the plugin part (UI and OctoPrint interaction) that might need some additional testing and development.  Let me know what you think.  I'd also be interested in seeing what the new *Firmware Checker* says about your firmware version.  That feature is extremely hard to test without user feedback.

Thanks for your post!

JasonLooseArrow, 2021-01-11:
This is very cool. I'm going to get on this and I'll let you know how it works! Thank you for the help and thank you for letting me work with the dev version.

JasonLooseArrow, 2021-01-11:
I am embarrassed to confess that I could not figure out how to download the plugin. I cut and paste into the ". . .from URL" part in the plugins manager, but when I do I get the message: Error: Could not install plugin (filename but no suffix) was neither a plugin archive nor a single file archive.

Can you open this picklejar for me? TIA!

FormerLurker, 2021-01-11:
Hmm, I guess I pasted it in wrong, sorry!

```
https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip
```

JasonLooseArrow, 2021-01-11:
No worries, I appreciate your help!

FormerLurker, 2021-02-06:
Life stuff still getting in the way of printing?  😆 Anyway, just checking back to see if there's been any changes.

JasonLooseArrow, 2021-02-06:
Yes. Pretty much, stuff is a mess. I upgraded up to 2.0.7. That went fast. Prints have dramatically improved and it looks like it's doing arcs, which is what it's supposed to do? But, the prints look so good, and I have so much else on my mind, I keep forgetting about using ArcWelder. So, there is that. Give me another week, I'll try to post some comparisons.

FormerLurker, 2021-02-06:
No problem Jason!  Take your time, and hopefully find some to relax and enjoy yourself.  Issues can wait!