FormerLurker/ArcWelderPlugin #115: Increasing fall in conversion accuracy in higher layers

FrenkK, 2021-01-09:
Hello :)
A rather large print of mine spaghettified recently for no obvious reason. Looking at the timelapse, nothing seems out of place until there is just a lot of pasta everywhere. This almost never happens, so I took a look at the ArcWelder gcode backimported in Cura and it looks like there were some problems in conversion: After about the 390th layer the extrusions start veering of course more and more sharply and it looks like some layers are either missing or there is an unwanted Z-move. To me, it looks like a bug, possibly a numeric error that accumulates. It might be worth looking into it :)
I attached the converted gcode, you can see the artefacts after the 390th layer. It is a quite simple mechanical part, all surfaces (except for support) should be either planar or circular and you can clearly see the outliers. Sorry, but I no longer have the original before conversion. I used dynamic layers (possibly a clue, looking at Z-moves) and tree support in latest Cura, the rest of the settings should be standard. Gcode is for Klipper.
Thanks for all your efforts :)
[CFFFP_Hrbtni_pokrov_manjsi.zip] (https://github.com/FormerLurker/ArcWelderPlugin/files/5790341/CFFFP_Hrbtni_pokrov_manjsi.zip)

FormerLurker, 2021-01-09:
What version of the plugin are you using?

FrenkK, 2021-01-10:
The latest available in Plugin Manager, 1.0.0.

FormerLurker, 2021-01-10:
OK, thanks.  I have reviewed your gcode, but it's very difficult to figure out what is going on without the original file.  Perhaps you could attempt a re-slice?

However, I have gotten other reports of issues with tree supports using version ArcWelder V1.0.0.  I  have added some major enhancements to the arc generation algorithm to detect and prevent any potentially bad arcs that you may want to try.  However, if you can first reproduce the issue by attempting to re-slice the original STL and post them, I could definitively test the fix myself.  Here is a link you can use to install the new version via the plugin manager (get more->from url):  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/a0f41e1ae75443805278b643ed9162bfcae17394.zip```

Thanks!

FormerLurker, 2021-02-06:
Any movement here?  Let me know if you were able to test the dev build, or if you are still having problems, thanks!

FrenkK, 2021-02-06:
Sorry to waste your time, I wasn't able to reproduce.

closed 2021-02-06 by FrenkK

FormerLurker, 2021-02-06:
Hey, no problem!  I'd rather have a couple false positives once in a while than a bunch of unreported errors.  Thank you for your post!