FormerLurker/ArcWelderPlugin #116: PreviewPictures are not visible with the "delete" Feature

Krieselwind, 2021-01-09:
Hello,
i have a problem. I use Octorpint with the Cura Plugin and OctoDash. If slice my files in Cura i can send it diretly to Octoprint. 

If I use "don't delete files after converting" i can see the Preview Image also on the new generated files. If I chose "delete files" the plugin didn't create the preview image :( 

[image]
[image]

FormerLurker, 2021-01-09:
Which version of the plugin are you using exactly?  Check the bottom right hand corner of the arcwelder tab.

Krieselwind, 2021-01-10:
hi, i found out, that it didn't depend on Cura. Sometimes i got the preview pictures and sometimes not. If i delete the generated AW file and generate a new one, the picture will be there...sometimes. Not easy to reconstruct the failur...

i use arc welder v1.0.0+u.bb71e8f

FormerLurker, 2021-01-10:
OK, I suggest upgrading to the devel branch, which has official support for the thumbnails when using the cura thumbnail plugin or the prusaslicer thumbnail plugin.  You can install it from the plugin manager (get more->from url...) with this URL: ```https://github.com/FormerLurker/ArcWelderPlugin/archive/a0f41e1ae75443805278b643ed9162bfcae17394.zip```

Let me know if that takes care of it.  Thanks!

FormerLurker, 2021-02-06:
Have you been able to verify that the dev version works with your thumbnail plugin?  I actually added a few more fixes to the most recent development branch..  Let me know.

Krieselwind, 2021-02-07:
Sorry for the late response, I test the new version in the last 10 days and I think it works :) 

many thanks  👍

FormerLurker, 2021-02-07:
Excellent!  I'll mark this as closed.  Enjoy the software!

closed 2021-02-07 by FormerLurker