FormerLurker/ArcWelderPlugin #117: Arc welder not creating arcs

KenLucke, 2021-01-10:
Should AW not be turning these segments into arcs?

[image]

[image]

FormerLurker, 2021-01-10:
I need to see the source and target gcode to determine this.  It also wouldn't hurt to see your arcwelder settings.

KenLucke, 2021-01-10:
Settings are default

[image]

Here are the files, before & after:
https://www.dropbox.com/s/8odecip8le4zssv/Crab%20Float.zip

KenLucke, 2021-01-15:
This seems to have solved itself, somehow.  See #122 .
(ain't computers fun?)

You can close it if you feel comfortable doing so,

FormerLurker, 2021-01-15:
@KenLucke, I'm going to leave this up for a bit since I would like to run the numbers on the bit you pointed out just as a sanity check.  I've been tweaking the algorithm a lot (it's a never ending process), and I think it's a worth seeing if I can explain this.

You've posted a lot in here, and i wanted to thank you for being so engaged!  I appreciate it.

FormerLurker, 2021-01-23:
Hey, going to close this since I can't seem to get to the source files anymore.  Let's call it a fluke :)

closed 2021-01-23 by FormerLurker

KenLucke, 2021-01-23:
Damn, I switched to a new laptop due to my old one needing a keyboard replacement (almost all the keys were doooooubbble and tripppppppple tapppppppping and I had to fix 3 out of 5 words I'd type, which was highly annoying) and was doing some cleaning up of my local dropbox space and deleted them.  I had ass/u/med you had downloaded them.

I agree with calling it a fluke unless it pops up again.