FormerLurker/ArcWelderPlugin #118: Using Arc Welder while automatically printing on file upload?

dgorbunov, 2021-01-11:
I am using Cura, where the Octoprint plugin will let you select if you want the print to automatically start after you upload GCODE. How can I use both this feature and Arc Welder so I don't have to manually start the print every time? Basically, how can Arc Welder 'interrupt' this process and modify the file before it gets printed? Is this even possible? I can't be the only one with this workflow.

dgorbunov, 2021-01-11:
Ah, see #98 and #46 . My bad

closed 2021-01-11 by dgorbunov

FormerLurker, 2021-01-11:
Try using the Cura Arcwelder plugin instead.  It's written by FieldOfView, and wraps the console version of arc welder for use with Cura.  Then you can just use the Cura Plugin for Octoprint to do the rest.  Enjoy!

FormerLurker, 2021-01-11:
OH, I see you found those issues.  I did add lots of options to ArcWelder, and am in the final stages of perfecting the next release.  There are lots of nice new things in there that you can't do with a slicer plugin, like firmware checking, statistics info, and some other stuff, but at the end of the day all you need is the gcode :)