FormerLurker/ArcWelderPlugin #119: Unable to change release channel

bill-orange, 2021-01-12:
Oddly, I can no longer use the release channel selection.  Uninstalling ArcWelder had no effect.  

Perhaps you are in the process of releasing a change?
[image]

FormerLurker, 2021-01-12:
It's a bug unfortunately.  I implemented the new per plugin release channels blind, before it was available, and sort of forgot I did that.  Then I made the mistake of deploying without testing it on the OctoPrint release candidate.  I'm going to talk to Gina to see if it can be fixed via an OctoPrint update.  Otherwise this will require a manual uninstall/reinstall as soon as the next version of ArcWelder is released (which will hopfully be soon, still working on testing the new build and dealing with some performance issues)

bill-orange, 2021-01-12:
thanks, Makes sense.

closed 2021-01-12 by bill-orange