FormerLurker/ArcWelderPlugin #120: Rounded Corners cause Problems + Missing Geometry on some layers

OddHero, 2021-01-12:
I am using the default Settings in AW and the latest Prusa Slicer 2.3.0 for gcode generation.
Printing on a CTC Bizer.

[image][image][image]

The Infill seems to be mostly correct, but the outline is really messed up.
When I noticed the wrong outline on the first Layer I stopped printing. Looking at the generated Code in [gcode.ws](https://gcode.ws) there are a few really weired artifacts as well as some layers where the outlines of some features are completely missing...
When my current Print is done ill look at reducing the maximum arc radius, but i think there is probably something else going on here.

[GCode_and_stl.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5803668/GCode_and_stl.zip)

FormerLurker, 2021-01-12:
I've added a workaround for this problem, which is a firmware limitation (MM_PER_ARC_SEGMENT=1.0 in firmware without newer options introduced in marlin 2.0).  See [this issue](https://github.com/FormerLurker/ArcWelderLib/issues/18) in the ArcWelderLib (contains the core algorithm) for details.   If you'd like to try it out, let me know and I'll send you a link to the appropriate devel version.  Thanks!

FormerLurker, 2021-01-12:
also gcode.ws is not capable of rendering gcode with g2/g3 commands.  Simplify 3d works well, and the pretty gcode viewer plugin does as well.  Here is a closeup of the gcode run through simplify 3d:

[image]

Long story short:  assume your gcode viewer cannot render g2/g3, unless you know for a fact that it can :)

OddHero, 2021-01-12:
> I've added a workaround for this problem, which is a firmware limitation (MM_PER_ARC_SEGMENT=1.0 in firmware without newer options introduced in marlin 2.0). See [this [...]

So it is possible that my printer doesnt support g2/g3 as well?
its a pretty old marlin firmware so maybe...
Anyway yes, i would like to try out the dev version :)

FormerLurker, 2021-01-12:
> So it is possible that my printer doesnt support g2/g3 as well?

It is possible.  I would try a very small cylinder as a test.  Download the pretty gcode viewer plugin to visualize the model to ensure it's correct.  Then try to print the first layer.  Alternatively, you could construct a simple G2/G3 command and send it to your printer as a test.  I believe I have an example script in the arcwelder plugin page on plugins.octoprint.org if you want to try that.

FYI, you may also want to slice using relative extrusion.  That way if your printer turns out not to support arc commands it will simply omit the curves rather than trying to jam extra filament missed by the arcs into the next move.  It should be obvious very quickly if arc commands are being ignored.

> its a pretty old marlin firmware so maybe...

Upgrade that puppy!  Trust me, things have only gotten better over the years.  It is a bit of a pain, but once you are up to date you can easily continue to stay current.  Just be sure to save your config file :)  If you don't upgrade now, plan on doing it at some point soon because you won't regret it (after you get it working that is, lol).

> Anyway yes, i would like to try out the dev version :)

Fantastic!  Here is the latest reasonably stable devel build that you can install via the plugin manager (OctoPrint Settings->Plugin Manger->Get More->From Url...):  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip```

That is not the most up to date version, but I've been working on an enhancement that isn't quite done yet (it's just a performance boost, not anything significant to the final gcode).  Please give me some feedback.  The UI is way different, so be prepared for that.

FYI, this version is actually somewhat safer than the existing version in that it has been shown to handle some edge cases better than the old version, while providing more compression and accuracy (it is minor, but still there).

Once it is installed, the thing I think you should look at is the new firmware checker, which will report any known issues if the version can be properly detected.  It might even be able to tell you if arcs are enabled or not (depends on the version).  Once you know if g2/g3 is working, try enabling firmware compensation, which will fix those corners but sacrifice a bit of compression.  Honestly, that is a must have feature if your firmware doesn't have some of the new fancy settings for arc generation.

Let me know how it goes!

OddHero, 2021-01-12:
Alright, i guess i have to do the firmware upgrade sooner than later 😄 

Send: G2 X30 I5 J10
Recv: ok
Recv: // echo: (line 107) Syntax warning: unrecognised command word 'I'
Recv: // echo: (line 107) Syntax warning: unrecognised command word 'J'
Recv: // echo: (line 107) Syntax warning: unsupported gcode command 'G2'

That explains why it wasnt working ^^
Thank you for helping with the fast responses 😄

FormerLurker, 2021-01-12:
@OddHero, lol!  I thought that was a strong possibility.  Please try installing the dev version and see if it tells you that arcs are not enabled:

[image]

I haven't gotten enough feedback on that feature to feel comfortable that it works.

OddHero, 2021-01-13:
[image]

Ohh yea Sailfish... its a mightyboard in there...
I am going to upgrade the firmware to run Klipper now, always wanted to see if i can get faster feeds out of the printer that way :)
I will try the Plugin again once i have that up and running.
Thanks again for the support 😉

OddHero, 2021-01-13:
Ohh and here is a verbose Log of the Firmwarecheck. Uploading that now, as i am not going to be able to test when changing the firmware.

[plugin_arc_welder.log](https://github.com/FormerLurker/ArcWelderPlugin/files/5805644/plugin_arc_welder.log)

[current.json](https://github.com/FormerLurker/ArcWelderPlugin/files/5805650/current.json.txt)

FormerLurker, 2021-01-13:
Awesome!! I was going to ask for you to run M115 and paste in the output, but that is (hopefully) in the log.  Good luck with your klipper migration!!!

FormerLurker, 2021-02-06:
Any update here?  Is this issue safe to close?