FormerLurker/ArcWelderPlugin #121: Plugin installed but stay in gray

mab18, 2021-01-13:
Hi, 
First of all, thanks you for this plugin and have a happy new year !

I installed the plugin, and installed too apt-get install -y python3-dev (I am under ubuntu)

[image]

It stay in gray (not 50 ;-) ) and I don't see any new function...

FormerLurker, 2021-01-13:
Try rebooting the pi (not just restarting OctoPrint) and see if that helps.  If not, uninstall and reinstall.  Let me know if that works.  If not, we'll continue trying to figure out what's happening exactly!  You may want to look at octoprint.log and see if it reports any errors while trying to start the plugin.

mab18, 2021-01-13:
I don't have a pi, I have a pc under ubuntu, just in case of.

FormerLurker, 2021-01-13:
> I don't have a pi, I have a pc under ubuntu, just in case of.

Ok, that shouldn't be a problem, unless there is some compatibility issue I'm not aware of.  Just make sure to reboot before trying anything else, but looking in the log beforehand is just fine.

mab18, 2021-01-13:
Here my plugin log : 
```
2021-01-12 18:25:09,250   /usr/bin/python -m pip --disable-pip-version-check uninstall --yes Arc-Welder
2021-01-12 18:25:10,909 > Uninstalling Arc-Welder-1.0.0+u.bb71e8f:
2021-01-12 18:25:10,909 > Successfully uninstalled Arc-Welder-1.0.0+u.bb71e8f
2021-01-12 18:25:11,018 > Cleaning up plugin arc_welder...
2021-01-12 18:25:46,366   /usr/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpbfUXxT/master.zip --no-cache-dir --user
2021-01-12 18:25:47,749 > Processing /tmp/tmpbfUXxT/master.zip
2021-01-12 18:25:48,986 > Requirement already satisfied: OctoPrint>1.3.8 in /home/octoprint/.local/lib/python2.7/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.5.2)
2021-01-12 18:25:48,987 > Requirement already satisfied: setuptools>=6.0 in /usr/lib/python2.7/dist-packages (from Arc-Welder==1.0.0+u.bb71e8f) (39.0.1)
2021-01-12 18:25:48,987 > Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.11.0)
2021-01-12 18:25:48,987 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.0)
2021-01-12 18:25:48,987 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.12.1)
2021-01-12 18:25:48,987 > Requirement already satisfied: pylru<2,>=1.2 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.2.0)
2021-01-12 18:25:48,987 > Requirement already satisfied: feedparser<6,>=5.2.1; python_version < "3" in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.2.1)
2021-01-12 18:25:48,987 > Requirement already satisfied: enum34<1.2,>=1.1.10; python_version < "3" in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.10)
2021-01-12 18:25:48,987 > Requirement already satisfied: Click<8,>=7.1.2 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (7.1.2)
2021-01-12 18:25:48,987 > Requirement already satisfied: pyserial<4,>=3.4 in /usr/lib/python2.7/dist-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.4)
2021-01-12 18:25:48,987 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.7.19)
2021-01-12 18:25:48,988 > Requirement already satisfied: websocket-client<1,>=0.57 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.57.0)
2021-01-12 18:25:48,988 > Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2020.8.7)
2021-01-12 18:25:48,988 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.1.1)
2021-01-12 18:25:48,988 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.9)
2021-01-12 18:25:48,988 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.4.21)
2021-01-12 18:25:48,988 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.1)
2021-01-12 18:25:48,988 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.5.0)
2021-01-12 18:25:48,988 > Requirement already satisfied: watchdog==0.10.4 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.4)
2021-01-12 18:25:48,988 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.11.2)
2021-01-12 18:25:48,988 > Requirement already satisfied: rsa==4.0 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (4.0)
2021-01-12 18:25:48,989 > Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.16.3)
2021-01-12 18:25:48,989 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.1)
2021-01-12 18:25:48,989 > Requirement already satisfied: flask<2,>=1.1.2 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.2)
2021-01-12 18:25:48,989 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.0)
2021-01-12 18:25:48,989 > Requirement already satisfied: frozendict<2,>=1.2 in /home/octoprint/.local/lib/python2.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.2)
[... cut ...]
2021-01-12 18:26:09,066 > Created wheel for Arc-Welder: filename=Arc_Welder-1.0.0+u.bb71e8f-cp27-cp27mu-linux_x86_64.whl size=1056285 sha256=8ed5b4f9161576f9153e4ee4887c6924063845c7c6327a9c45e63421b61211fa
2021-01-12 18:26:09,066 > Stored in directory: /tmp/pip-ephem-wheel-cache-jpuwKS/wheels/d5/83/43/2e7e73a875bc08ede87044aa3c0be11de3dfe1a0a82b6c10cd
2021-01-12 18:26:09,067 > Successfully built Arc-Welder
2021-01-12 18:26:09,067 > Installing collected packages: Arc-Welder
2021-01-12 18:26:09,068 > Successfully installed Arc-Welder-1.0.0+u.bb71e8f

```

FormerLurker, 2021-01-13:
What we really need is octoprint.log.  also, in the future please post logs to gist.github.com, and link to them in this issue.  Things get too long really fast otherwise.  Thanks!

mab18, 2021-01-13:
[The log as requested](https://gist.github.com/mab18/3685e5412d80d6d7d21f2453d1f2a329#file-log-octoprint-arc_welder)

FormerLurker, 2021-01-13:
I need octoprint.log, not plugin_arcwelder.log please.  This one:

[image]

Thanks!

mab18, 2021-01-13:
The link is this octoprint.log file, but extract. Do you need the entire file ? There is no error elsewhere in the file.

FormerLurker, 2021-01-13:
The entire file please.  If it is really really big, you could delete the existing log, restart octoprint, then post that log.  Thanks!

FormerLurker, 2021-01-13:
FYI, the error you posted is known, and won't prevent ArcWelder from working.

mab18, 2021-01-13:
I added entire file in the link

FormerLurker, 2021-01-13:
OK, so it doesn't look like OctoPrint itself is having a problem loading the plugin.  

IF arc welder settings are showing up here:

[image]

Try this:

1.  Clear out the plugin_arcwelder.log file by going to the Arc Welder settings and clicking this button:

[image]

2. Enable verbose logging for the arc_welder.__init__ module like so:

[image]

3. Verify that your logging settings now look like this:

[image]

4. Save the settings changes.  

5. Restart Octoprint.

6. Go back into the arc welder settings and download the log like so:

[image]

7. Remove the verbose logging to prevent future issues:

[image]

8. Save your changes.

9. Post the log file you downloaded.

If your settings are NOT showing up there, let me know.

mab18, 2021-01-13:
Oh... You know what... ? It works... :-)) But I don't know why. :-//

I had already rebooted my machine, uninstalled/reinstalled...

So , Thank you so much for your help
[image]
[image]
I have now to learn how to do with it ;-)

Thanks again 👍

FormerLurker, 2021-01-13:
Awesome!!!!  If you want you can try the devel version, which needs some testing.  There are a lot of cool new features, and a firmware checker.  Let me know and I can send you a link you could use to install via the plugin manager.

mab18, 2021-01-13:
It's ok for me !

FormerLurker, 2021-01-13:
Cool, you can install via the plugin manager (get more->from url...) ```https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip```

If you have problems installing, you can always uninstall it and reinstall the stable release.  Also, I recommend installing 'pretty gcode viewer' which can visualize models with G2/G3 commands (your slicer likely cannot, nor can most viewers).  I welcome all feedback!

KenLucke, 2021-01-15:
> If you have problems installing, you can always uninstall it and reinstall the stable release. Also, I recommend installing 'pretty gcode viewer' which can visualize models with G2/G3 commands (your slicer likely cannot, nor can most viewers). I welcome all feedback!

I thought that Pretty Gcode was "verboten" after the OP v1.5.x update due to some conflicts?  I admit, I haven't been following the PGC thread, so is it back in grace?  I liked it because it did what the OP authors said could not be done with the built-in gcode viewer:  handle huge files.

FormerLurker, 2021-01-15:
@KenLucke, it must have been fixed, or partially fixed at least, because I have been using it since my pull request was accepted (adding G2/G3 support).  Also, even before the PR (which was admittedly very recent), I was using it with some success while developing the new feature.

mab18, 2021-01-16:
Hi,
After install of this version, I have a fatal error : 
[image]
[image]

KenLucke, 2021-01-16:
As I am sure the author will tell you, it is best to open new, different issues in a, you guessed it, "New Issue" rather  than a  continuation of a different one.  This is the same issue as #124

FormerLurker, 2021-01-16:
@mab18, please hop over to issue #124, as @KenLucke  suggests.  It looks like your settings are not loading properly for some reason.

closed 2021-01-18 by mab18

mab18, 2021-01-18:
Next to my last post, I close the issue.