FormerLurker/ArcWelderPlugin #122: No issue, just an example

KenLucke, 2021-01-15:
Left: w/Arc Welder (done under Octoprint... the Cura extension always crashes for me)
Right w/o Arc Welder

[image]

FormerLurker, 2021-01-15:
Wow!  Thank you for posting this :)  Would you mind posting your settings?  Also, can I use this image?

KenLucke, 2021-01-15:
> Wow! Thank you for posting this :) Would you mind posting your settings? Also, can I use this image?

Slicer or AW settings?  AW is just default.

FormerLurker, 2021-01-15:
Just defaults!  That is nuts.

KenLucke, 2021-01-15:
> Just defaults! That is nuts.

Yeah, I'm fairly impressed.

FormerLurker, 2021-02-06:
Ok, going to close this out since I'm doing some housekeeping, but I am going to save this image and use it in the plugin repo when I drop the next release (that is if I can ever fix all the bugs you find, LOL)

closed 2021-02-06 by FormerLurker