FormerLurker/ArcWelderPlugin #123: One nitpicky detail

KenLucke, 2021-01-16:
Some OSs do not deal well with double file extensions (i.e., filename.something.ext

You may want to change the default new filenae designatore ".aw" to "_aw" or "-aw"

FormerLurker, 2021-01-23:
Hey @KenLucke , can you let me know what OSs have an issue with this and what exactly happens?  I just want to understand the problem a bit better before I make any changes.

KenLucke, 2021-01-23:
I just recall that I've used some OSs in the past that required only a single dotted file extension, otherwise they read the first ".something" as the extension.  It's not a big deal, just change the default in the settings to an underscore and no one will ever be the wiser.  It's not any code change, just a settings-default-out-of-the-box change.  I just changed mine in the settings and it's fine.

FormerLurker, 2021-01-30:
So I did a bit of research, and aside from some very old MacOs (no longer supported) double extensions seem to be fine in all OSs that ArcWelder supports (Windows, Linux flavors, MacOs), so I think I'll just leave it the way it is right now.

Good suggestion here, but it seems things have changed regarding compatibility for this kind of file.

Please keep these recommendations coming if you have other ideas though!  You've got a good track record of supplying winning ideas!

closed 2021-01-30 by FormerLurker

KenLucke, 2021-01-30:
LOL, all it would have taken was changing the default showing in the settings dialog on plugin install from ".aw" to "_aw",  I didn't mean for it to become that big of a deal.