FormerLurker/ArcWelderPlugin #124: RC Barfs on file

KenLucke, 2021-01-16:
I just installed the https://github.com/FormerLurker/ArcWelderPlugin/archive/2dedd1b9dfa5da4e75e968107473a8113369e709.zip version, and tried to convert it.  Multiple times.   Failed with an error every time. Rolled back to v1.0.0, and it converted it just fine.
 
 Log is short, so I'll post  it here:
 ```
2021-01-16 01:11:58,078 - arc_welder.__init__ - ERROR - Could not process file manually.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 317, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 1304, in add_file_to_preprocessor_queue
    if "arc_welder" in metadata:
TypeError: argument of type 'NoneType' is not iterable
2021-01-16 01:12:18,360 - arc_welder.__init__ - ERROR - Could not process file manually.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 317, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 1304, in add_file_to_preprocessor_queue
    if "arc_welder" in metadata:
TypeError: argument of type 'NoneType' is not iterable
2021-01-16 01:19:20,206 - arc_welder.__init__ - ERROR - Could not process file manually.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 317, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 1304, in add_file_to_preprocessor_queue
    if "arc_welder" in metadata:
TypeError: argument of type 'NoneType' is not iterable
2021-01-16 01:20:47,647 - arc_welder.__init__ - ERROR - Could not process file manually.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 317, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 1304, in add_file_to_preprocessor_queue
    if "arc_welder" in metadata:
TypeError: argument of type 'NoneType' is not iterable
2021-01-16 01:21:32,838 - arc_welder.__init__ - ERROR - Could not process file manually.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 317, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 1304, in add_file_to_preprocessor_queue
    if "arc_welder" in metadata:
TypeError: argument of type 'NoneType' is not iterable
2021-01-16 01:24:08,075 - arc_welder.__init__ - ERROR - Could not process file manually.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 317, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_arc_welder/__init__.py", line 1304, in add_file_to_preprocessor_queue
    if "arc_welder" in metadata:
TypeError: argument of type 'NoneType' is not iterable

```

mab18, 2021-01-18:
As I said, this version is dramatic : plugin is crashed and is not in the list of plugins : cannot be installed and cannot be uninstalled... stable version install doenst correct the problem 🤔

mab18, 2021-01-18:
I deleted the error I had : I note that service not well restart when  I click on restart octoprint. I need to see why. When I send the "service octoprint restart", it's ok. I have to make a print tomorow, so I will see if everything is ok with with version.
Sorry for my false positive error and sorry for added it to bad issue. ;-)

FormerLurker, 2021-01-18:
Np @mab18!

@kenlucke, looking into your issue.  Will report back tomorrow

mab18, 2021-01-18:
I have to print a gearwheel now, I send you my results ;-)

FormerLurker, 2021-01-18:
@KenLucke, try installing this build:

```https://github.com/FormerLurker/ArcWelderPlugin/archive/cb5e62706269bce0333017d095ace93ba815535e.zip```

Let me know if that fixes your issue.

KenLucke, 2021-01-18:
> @KenLucke, try installing this build:
> 
> `https://github.com/FormerLurker/ArcWelderPlugin/archive/cb5e62706269bce0333017d095ace93ba815535e.zip`
> 
> Let me know if that fixes your issue.

Yes, that seems to have solved it.  Thank you.

KenLucke, 2021-01-18:
BTW, how do I go about getting a Marlin version into the repository you access for compatibility check?  The firmware version of Marlin I'm using does have everything enabled, but is not  recognized.  I  realize it's just cosmetic and a warning, but others might be concerned (and there now a LOT of people using Nic Wiilson's branch of Marlin 2.0.7, and I've customized it a little further.

FormerLurker, 2021-01-18:
Send an m115 command and paste the results in here.  Also, a snap of the warnings would be great

KenLucke, 2021-01-18:
As for the warning, it's just the standard one you show  when the firmware is not recognized. 

Any Marlin 2.0.[4+].x_)NIC will have these same features enabled.

```
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.7.7B_NIC (Dec 31 2020 13:09:28) SOURCE_CODE_URL:https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0 PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR10 V2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:0
Recv: Cap:AUTOREPORT_SD_STATUS:1
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok P15 B7
Send: M155 S2
Recv: ok P15 B7
Send: M876 P1
Recv: ok P15 B7
```

FormerLurker, 2021-01-18:
It is reporting marlin with arcs enabled, right?  The issue with that version scheme is that it is not semantic.  I will have to write a special routine to handle it

KenLucke, 2021-01-19:
> It is reporting marlin with arcs enabled, right? The issue with that version scheme is that it is not semantic. I will have to write a special routine to handle it

Aw, if it's just for me, don't bother... I know it's not affecting anything.

FormerLurker, 2021-01-19:
Is this your fork?  If so, consider just adding a space after the version number.

KenLucke, 2021-01-19:
> Is this your fork? If so, consider just adding a space after the version number.

Not mine, but I'll let Nic know.

FormerLurker, 2021-01-19:
Thanks!  I will run the existing version number through the parser to see what happens.  Also, there may even be another field that can be used.