FormerLurker/ArcWelderPlugin #125: Klipper crashes with converted file

DroneMang, 2021-01-17:
Hi,

I have been using arcwelder for some time and overall I would say I have had no issues, this converted file crashes the printer, causing Klipper to give an MCU shutdown, timer too close.

Here is the unconverted file and converted.  

I use SolidWorks for my CAD, and when I export I use the very highest resolution settings, makes large STL/3MF files, but works well usually, perhaps I need a more coarse file?  I am using Simplify3D for the slicer.

Also, I have this set in Klipper:

[gcode_arcs]
resolution: 0.05 # 1.0 default

[File.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5826619/File.zip)

FormerLurker, 2021-01-17:
I will take a look.  Whay version are you running?

DroneMang, 2021-01-18:
Of arcwelder?  Whatever the newest that octoprint updates to, looks like 1.0.0+u.bb71e8f?  Should I have arcwelder set to the stable release channel?  Thanks!

FormerLurker, 2021-01-18:
Ok, so I've been researching the error you observed, and it seems like it's typically a communication problem.  Obviously you're not having this trouble on other files, so perhaps something else is going on?  Can you take a look at the terminal output when you are trying to print that file?  Maybe that will be illuminating.

DroneMang, 2021-01-18:
If I select dev from within octoprint, will it use the 1.0.1rc1.dev2 version or do I need to manually install that?

FormerLurker, 2021-01-18:
You would have to paste in an archive url.  I can send the most recent link when i get back.

However, if you have time, see if you can replicate the error and capture the terminal output (the octoprint terminal tab).  100 lines or so should be plenty

DroneMang, 2021-01-19:
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 G90*17
Recv: ok
Send: N2 M83*26
Recv: ok
Send: N3 M106 S41*81
Recv: ok
Send: N4 M140 S60*87
Recv: ok
Send: N5 M190 S60*91
Recv: ok
Send: N6 M104 S240 T0*33
Recv: ok
Send: N7 M104 S240 T1*33
Recv: ok
Send: N8 M109 S240 T0*34
Recv: ok
Send: N9 M109 S240 T1*34
Recv: ok
Send: START_PRINT
Recv: ok
Send: N10 M105*22
Recv: ok B:59.7 /60.0 T0:238.9 /240.0
Send: N11 T0*10
Recv: // Extruder extruder already active
Recv: ok
Send: N12 G1 Z0.200 F480*23
Recv: ok
Send: N13 G1 X38.161 Y6.781 F21000*123
Recv: ok
Send: N14 G11*44
Recv: ok
Send: N15 G3 X42.308 Y3.013 I4.339 J0.611 E0.26249 F6600*13
Recv: ok
Send: N16 G1 X48.514 Y3.010 E0.2683*84
Recv: ok
Send: N17 G3 X49.029 Y11.737 I-0.032 J4.381 E0.56980*112
Recv: ok
Send: N18 G1 X48.310 Y11.790 E0.0312*107
Recv: ok
Send: N19 G2 X75.971 Y136.701 I336.760 J-9.057 E5.56409*79
Recv: ok
Send: N20 G2 X148.399 Y242.460 I309.032 J-133.957 E5.57479*121
Recv: ok
Send: N21 G2 X255.093 Y313.5 I236.616 J-239.731 E5.57479*118
Recv: ok
Send: N22 G2 X380.603 Y339.531 I129.907 J-310.762 E5.57479*119
Recv: ok
Send: N23 M105*22
Recv: ok B:59.5 /60.0 T0:237.5 /240.0
Send: N24 G2 X506.75 Y316.785 I4.395 J-336.796 E5.57479*78
Recv: ok
Send: N25 G2 X615.263 Y248.555 I-121.771 J-314.070 E5.57479*84
Recv: ok
Send: N26 M105*19
Recv: // Klipper state: Shutdown
Recv: ok B:59.6 /60.0 T0:237.5 /240.0
Send: N27 G2 X690.424 Y144.723 I-230.278 J-245.814 E5.57479*82
Recv: // MCU 'mcu' shutdown: Timer too close
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: Timer too close
Changing monitoring state from "Printing" to "Cancelling"
Send: END_PRINT
Recv: ok
Recv: // MCU 'mcu' shutdown: Timer too close
Changing monitoring state from "Cancelling" to "Operational"
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: Timer too close
Recv: ok

DroneMang, 2021-01-20:
Where should I install python 3 from?  I have your command from your documentation, but there is an Octoprint link as well: https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

FormerLurker, 2021-01-20:
I think you will still need to upgrade to python 3.  You can use the script you linked to.  Best to run that befort installing any plugins (less to go wrong then).  The command I gave installs the development tools, but the script may install that too (i suspect it does).  If you can install arc welder after the upgrade, then you don't have to run the command i gave.

Also, I still need to look into your terminal log.  Not been able to work on this at all since my last message, but will have time Friday at the latest.  I also forgot to send you the latest install url, so here it is:

https://github.com/FormerLurker/ArcWelderPlugin/archive/cb5e62706269bce0333017d095ace93ba815535e.zip

Install via the plugin manager (get more->from url...)

FormerLurker, 2021-02-06:
I can't seem to figure this out.  Looking at some other klipper issues now, so maybe it will help.  Just wanted to let you  know it hasn't fallen off my radar.