FormerLurker/ArcWelderPlugin #126: Slicing file removes retractions and adds features for UM3 printing

seantapscott, 2021-01-19:
Using Arc Welder inside Cura 4.8.0 (https://github.com/fieldOfView/Cura-ArcWelderPlugin/issues/8) . It created weird results on the converted gcode file.  

The WheelHolder model is supposed to be simple semi-circle arcs of 2mm thickness with filleted edges.  Using arc Welder created straight lines across the opening of the semi-circles.  

[image]

Original model:
[WheelHolders.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5838329/WheelHolders.zip)

Arc Welder converted file and unconverted file:
[UM3E_WheelHolders_43g_arcs.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5838338/UM3E_WheelHolders_43g_arcs.zip)

Simplify3D shows an accurate gcode preview, but UM3 prints it with straight lines across the front, and Cura also previews the gcode in this way.  

On a side note, I did open a ticket with FieldOfView's plugin github repository, but we're not too sure what the root cause is here.  He suggested that perhaps the UM3 is not interpreting the arcs correctly. I could find no fault with the converted gcode file myself.

FormerLurker, 2021-01-19:
It could definitely be a G2/G3 interpretation issue within the ultimaker firmware.  I did notice that the skirt looks messed up, even on the original file, so there could also be a problem with the slicing.  Do you have a photo of what your printer did with the converted gcode?  It should be pretty easy to tell if arcs are being interpreted as straight segments (basically G1 commands).

That being said, simplify is generating a preview that looks identical to the source gcode file.  The OctoPrint pretty gcode viewer is also displaying the file correctly:

[image]

I would recommend posting on the ultimaker forums and asking if G2/G3 codes are supported.  If they are not, I would recommend to them that they implement these gcodes.  However, looking at [what I assume is the firmware running on your printer](https://github.com/Ultimaker/Ultimaker2Marlin), it seems that these codes are supported.  Maybe they need to be initialized, or perhaps you are running some other firmware?  I've never used an ultimaker before, so I don't have much knowledge there.  You could try sending an M115 command and pasting the output in here.

Thanks!

Also, doesn't ultimaker

seantapscott, 2021-01-19:
The skirt is actually correct.  Ultimaker 3 (not 2 like the firmware you listed) has two nozzles with an X-axis offset, so I'm guessing the gcode handles it like that. The two lines are both extruders drawing a skirt. 

Anyway, I think we can close this.  Ultimaker's not interpreting the arcs correctly.

closed 2021-01-19 by seantapscott

FormerLurker, 2021-01-19:
Ok, since this is probably a downstream issue, I agree.  However, if you could send your printer an M115 command, I'd still be interested in seeing the results.  That would allow me to update the new firmware checker in v1.1 to display Ultimaker firmware info for your version to prevent this issue in the future (at least for the OctoPrint plugin users).  Also, please still consider asking Ultimaker to support G2/G3 commands in their V3 firmware.

Thanks!

seantapscott, 2021-01-19:
Ultimaker 3 doesn't easily allow direct M115 commands.  Not really a console, as far as I know. FieldOfView is going to try to improve the Preview option if he can, but not sure what else'll happen.