FormerLurker/ArcWelderPlugin #127: Gcode lines being changed

opjose, 2021-01-20:
I ran into a situation where Arc-Welder is changing straight wall lines repeatedly.
In layer 519 or so in my print the straight line wall was abruptly changed into an angle.  Arc-Welder is set to the default settings.
[image]
[image]

opjose, 2021-01-20:
Here are the two G-code files, one after Arc Welder processing...
[TT_ATX_case_v5.gcode.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5840542/TT_ATX_case_v5.gcode.txt)
[TT_ATX_case_v5.aw.gcode.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5840545/TT_ATX_case_v5.aw.gcode.txt)

FormerLurker, 2021-01-23:
Have you actually tried printing this?  The file looks ok in simplify 3d and the pretty gcode viewer plugin.  I do see what's throwing the OctoPrint gcode viewer off though, but I'm pretty sure it should print OK.  I'm going to see if I can make some changes to the algorithm to make this less likely, but I don't think it is a problem per se.

opjose, 2021-01-24:
No I did not try printing it fully. I started to print the part and noticed the incorrect wall in the preview, so I stopped the print and re-printed the unprocessed version.

I thought to bring it to your attention in case this was something that needed review.

Thanks for responding!!!  You have created the single most essential plugin IMHO.

THANK YOU!

FormerLurker, 2021-01-24:
I couldn't do it without people like you, who report issues, so thanks!  Again, technically the gcode is correct, and it should print ok (at least in marlin), but it illustrated a case scenat i havent covered, which will be a good thing to implement (combination of colinear points into a single line)

enz1ey, 2021-02-06:
I just experienced what I believe to be this same issue. Not quite sure yet, I'll have to troubleshoot a bit later today to get some information and screenshots.

I installed Arc Welder using the OctoPrint plugin manager, then clicked the Arc Weld button next to a print already in the file list in OctoPrint. I then started printing the aw file, and instead of a normal rectangular box, it started printing a rectangle that was rotated a few degrees clockwise on the XY axis, then started printing another rectangle rotated equally in the counter-clockwise direction.

FormerLurker, 2021-02-06:
Yes, I would like to see some snapshots and the source/target gcode files please.  I'd like to visualize the files and run them through the newer version to see if anything has changed.