FormerLurker/ArcWelderPlugin #128: Arc-Welder errors when reloading OctoPrint (or a plugin) after an update

b-morgan, 2021-01-22:
My RPi is always on but my printer is only on when I'm printing. If OctoPrint has an update (i.e. 1.5.2->1.5.3 ) or a plugin has an update, Arc-Welder throws two errors if the printer is off (i.e. not connected). One is "Error Loading Pre-Processing" and the other is an error about the Firmware.

I'm using OctoPi 0.18.0, OctoPrint 1.5.3, Python 3 on an RPi 3B connected to a Lulzbot TAZ 6.

I would prefer that these error messages be suppressed if the printer isn't connected.

FormerLurker, 2021-01-22:
Which version of arc welder?

b-morgan, 2021-01-22:
(I clicked on About looking for it and it wasn't there so I forgot to include it)

Arc-Welder (1.0.1rc1.dev2+u.2a95fab)

FormerLurker, 2021-01-22:
Thx, i will add the version to about as well

FormerLurker, 2021-01-22:
I'm not able to replicate this with the latest devel version, so perhaps it is fixed?  Here are the steps I took:

1. Disconnect my printer and uncheck auto connect.
2. Reboot and ensure the printer remains disconnected.
3. Upgrade OctoPrint to 1.5.3 and reboot (no error)
4. Upgrade the firmware updater plugin and reboot (no error).

Can you perhaps try installing this latest devel release and see if you encounter an issue:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/cb5e62706269bce0333017d095ace93ba815535e.zip```

Let me know how it goes please, and as always thank you VERY much for your report!

b-morgan, 2021-01-22:
I installed the latest devel release and got the two errors. Since a reboot is required, I figured the errors came from the old version.

I installed the latest devel release again and again got the two errors. I am assuming that these errors came from the new version.

Here is a screenshot. You can't read the error messages because they are behind the Please Reload dialog, but I'm pretty sure they are the same ones as in the OP.

[image]

FormerLurker, 2021-01-22:
Ok, i will dig in.  I have not seen thise messages.  Development is hard, lol

KenLucke, 2021-01-24:
> Ok, i will dig in. I have not seen thise messages. Development is hard, lol

I can confirm this bug.  It only happens on reload after a restart of OP, and normal function seems to be fine after OP finishes reloading.

[image]

1. Restart (not reload) OctoPrint
2. Click the "Reload Now" button when it appears
3. Voilà

[octoprint(2).log](https://github.com/FormerLurker/ArcWelderPlugin/files/5862930/octoprint.2.log)

FormerLurker, 2021-01-24:
I tried 10 times, and it happened once.  I believe the core issue is that the onDataUpdaterReconnect event occurs before the blueprints work (404 error calling them apparently, but I can't be sure since the js gets reloaded).  This isn't convenient, and the fix will prevent errors that are legit from being shown, so as a compromise I'm adding the error to the console log instead of a popup.  I'm testing the fix now, but this is difficult for me to replicate.

KenLucke, 2021-01-24:
Odd, it happens 100% of the time for me.  Nothing (literally) shows in the Arc Welder debug log.

FormerLurker, 2021-01-24:
Yeah, because the error is that the browser is unable to communicate with the backend, so nothing ever gets to the python code.  This could be timing related (when is the reload button clicked), or maybe connection related (I'm hardwired).  I was recording my screen when trying to replicate, and it was about 10 tries before I saw it.  I have tried the same number of times on the fix, and got no errors, but you may want to try it since you're not having problems replicating the issue.

FYI, issue replication is the #1 most difficult thing to deal with.  If I see a problem when debugging/testing, I've probably fixed it.  It's the errors I don't see for whatever odd reason that give me trouble.

b-morgan, 2021-01-24:
I get some kind of firmware error on my TAZ 6 because I'm running a "bugfix" version of Marlin 2.0. Has that got something to do with why I have the error 100% of the time?

FormerLurker, 2021-01-24:
What is the error?  Screenshot?

b-morgan, 2021-01-24:
[image]

FormerLurker, 2021-01-24:
Does the 'known issue' message not explat this adequately?  If not, any suggestions on how to improve it, or add something else?  You can disable the checker in the settings if it annoys you.

I have no idea why the bugfix branches don't include a base version number.  The version numbering for the bugfix branch doesn't give any info about what version it is based on unfortunately.  Maybe i will open an issue in marlin for this.

b-morgan, 2021-01-24:
When I said "some kind of firmware error" it was because I didn't remember what the error was... I understand the error and it doesn't bother me because arcs are working fine. The errors I'm getting when I update with the printer turned off (the original post) are annoying but are not preventing me from printing. I reported it because I don't like to be conditioned into ignoring errors because one day, one of the messages I'm ignoring will be a different error that I shouldn't have ignored :smile:

FormerLurker, 2021-01-24:
@b-morgan, you are absolutely correct, and I don't want to give you the mistaken impression that I don't think this is a real and serious issue!  You have consistently reported problems you have discovered, which have helped me greatly to improve my software over the years.  KEEP IT UP PLEASE!  I need more people like you and @KenLucke.

So, regarding your OP, please try this build and let me know if the popup error problem has been solved:

```https://github.com/FormerLurker/ArcWelderPlugin/archive/61ac54ea37631fb7cbc3ba05d31ed0e7362d830b.zip```

This can be installed in the usual way via plugin manager->get more->from url...

I know you know that, but other users may be lurking in here waiting for a fix :)

Also, I am truly interested in getting some suggestions to make the firmware checker warnings (errors are in red) less irritating.  I don't want to train anyone to ignore errors!  Maybe some kind of -dismiss- button for warnings, but not for errors?  I'm not sure what the best way to handle that is.

KenLucke, 2021-01-24:
Will be a while before a printer is free to try the new build.

b-morgan, 2021-01-25:
Down to one error message (from two)...

[image]

FormerLurker, 2021-01-25:
Drat!  Must have missed one.  Will get it and post back...

FormerLurker, 2021-01-25:
Try this version:

```https://github.com/FormerLurker/ArcWelderPlugin/archive/1ccf7f219c1166432bc7b9a9b20e32ba29516459.zip```

FYI, I did verify that this is a different error than the two I prevented.  This further illustrates that this is indeed a race condition of some kind, else all of the errors should trigger every time for everyone.  I would not expect to be unable to contact the sever after receiving the onDataUpdaterReconnect event, but it it definitely possible that I don't understand it well.

Since I seem to have unusual difficulty triggering this, consider trying 3 or 4 times to catch more/rare issues.  If not, I understand.

Thanks!

b-morgan, 2021-01-25:
I installed this one twice and the error message is gone.

FormerLurker, 2021-01-25:
Awesome, thank you very much!

KenLucke, 2021-01-26:
I am still getting the two error messages originally noted.  The key to getting them is to click the "Reload" button as soon as it appears after a restart.

FormerLurker, 2021-01-26:
Can you take a snapshot of the error so i can see what it is?

KenLucke, 2021-01-26:
Same one(s) as above ^
[image]

This is using the latest URL install you had above (https://github.com/FormerLurker/ArcWelderPlugin/archive/1ccf7f219c1166432bc7b9a9b20e32ba29516459.zip)

KenLucke, 2021-01-26:
Well, NOW, I can't get it to do it.  Weird.

FormerLurker, 2021-01-26:
Probably cached js somehow, i guess.  I searched the code for those messages, and they are not producing pop-ups anymore.

KenLucke, 2021-01-26:
> Probably cached js somehow, i guess. I searched the code for those messages, and they are not producing pop-ups anymore.

Yeah. it stopped doing it after I reloaded a half a dozen times.

FormerLurker, 2021-01-30:
Ok @KenLucke  and @b-morgan, I've released this fix in v1.1.0rc1.  If either of you would like to help test that, you can leave feedback [here](https://github.com/FormerLurker/ArcWelderPlugin/issues/134).

As usual, thank you for pointing this issue out and sticking with it all the way to the resolution!  Much appreciated.

closed 2021-01-30 by FormerLurker