FormerLurker/ArcWelderPlugin #129: Processed gcode has detrimental effect on small features using default parameter values

brettvitaz, 2021-01-23:
Thanks for the plugin! I have just started using it and am having some issue with small circular features.

I have not adjusted the parameter values from default.
Resolution = 0.05mm

The result of the processed gcode causes deformed posts on prints. Here is an example of unprocessed and processed output:
[image]

Here is the gcode used:
[gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5861072/gcode.zip)

Printer: Prusa MK3S
Slicer: PrusaSlicer 2.3

Is this something I should fix by adjusting the resolution, or is this some kind of issue that I should expect the plugin to handle?

FormerLurker, 2021-01-24:
Well, this is firmware related.  I have posted a pull request to fix this in the mk2/mk3 firmware github.  However, i wouldn't assume it will be accepted.  I have added a new feature to the devel Branch to deal with this shortcoming called firmware compensation (in addition to a bunch of other things).  Would you be willing to give it a go and see if it solves your issue?

brettvitaz, 2021-01-24:
Sure, I'll try out the changes. I have installed the plugin from the Devel branch and will post an update with the results.

FormerLurker, 2021-01-24:
excellent, thanks!  I think the default firmware compensation settings should work well for you:

[image]

Also, I'm interested in what the firmware checker says about your printer:

[image]

brettvitaz, 2021-01-24:
I have set the firmware compensation to your suggested values. The output is much improved and is now similar to the unprocessed version.

Here is the reported Firmware Compatibility:
[image]

FormerLurker, 2021-01-24:
You may want to consider posting an issue to the PrusaFirmware repository.  I think it would be good if they knew how many people are affected by the shortcomings of their current g2/g3 implementation.  If enough people are having problems like your, perhaps some changes will be made?

brettvitaz, 2021-01-24:
I will post an issue with PrusaFirmware. Does the same issue exist in Marlin? I have some Ender 3 printers running the latest Marlin release. I can check those some time next week.

FormerLurker, 2021-01-24:
@brettvitaz, the latest version of marlin has much enhanced arc interpolation capabilities.  Depending on the settings, you won't have any issues on the latest marlin.

brettvitaz, 2021-01-24:
Great! Thanks again.

closed 2021-01-24 by brettvitaz

brettvitaz, 2021-01-24:
Using the settings you suggested, I printed a part this morning and noticed significant straightening of the arc segments.

I'm not sure if they show up well in this photo, but here are the results:
[image]

FormerLurker, 2021-01-24:
You can increat min arc segments.  Marlin's default is 24, but you won't get much, if any, compression like that.  The firmware compensation is a hack, unfortunately, so the only real fix is to update the firmware.

FormerLurker, 2021-01-24:
When i said you won't get much compression, i meant for small arcs, like your example.  Large pieces will perform better.