FormerLurker/ArcWelderPlugin #130: Noticeable seam

appleboblin, 2021-01-24:
Thank you for making the plugin! I've just started 3d printing around 2-3 weeks ago.
I have tuned my machine so that the seam are not as visible. When I tried out the plugin, I noticed that the seam became very noticeable.

Here is an example of the seam:
[image]

Printer: Ender 3 V2
Slicer: PrusaSlicer 2.3, Cura 4.8
Firmware: Klipper

Are there anything that I can do to reduce the seam? I've already tuned pressure advance, do I need to tune pressure advance with the plugin enabled?

FormerLurker, 2021-01-24:
Can you attach your before and after gcode so i can make sure it isn't a problem with the algorithm?  Thanks!

appleboblin, 2021-01-24:
I have deleted the cylinder files not long ago from octoprint, not sure how to recover it. However, I do have another print with both files. It also have noticeable seam.
[image]

Here are the g codes.
 [Before](https://github.com/FormerLurker/ArcWelderPlugin/files/5862803/CE3PRO_bltouch_adapter_5_fillets_etc.gcode.zip)
[After](https://github.com/FormerLurker/ArcWelderPlugin/files/5862806/CE3PRO_bltouch_adapter_5_fillets_etc.aw.gcode.zip)

I can reslice the cylinder, process it and reprint it if needed.

appleboblin, 2021-01-24:
This file was sliced with Cura and the cylinder was sliced with PrusaSlicer

FormerLurker, 2021-01-24:
K, if you could slice and reprocess the cylinder, that would actually be better.  It will be much easier to review the gcode, which needs to be done by hand at great expense, lol.

Also, which version of klipper are you running (Sending M115 should work well to determine this)?

appleboblin, 2021-01-24:
FIRMWARE_VERSION:v0.9.1-153-g64097cb0 FIRMWARE_NAME:Klipper
[cylinder2_0.2mm_PLA__44m.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5862831/cylinder2_0.2mm_PLA__44m.gcode.zip)
[cylinder2_0.2mm_PLA__44m.aw.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5862833/cylinder2_0.2mm_PLA__44m.aw.gcode.zip)

FormerLurker, 2021-01-24:
I can't see anything obviously wrong with the welded gcode.  I made sure the wipe retractions added up to the proper amount on the first few layers, and did an overlay of the original and welded files:

[image]

As you can see, they are extremely close.  Here is what simplify shows with the 'retractions' box ticked for the welded code:

[image]

Note that the 'blobs' there are just simplify's retraction indicator.  However, it does illustrate where your seam is appearing nicely, almost like retract isn't working.  Have you tried printing the non-welded code to ensure the seam doesn't appear?  Here it is with that turned off:

[image]

I think I need to look into Klipper's arc interpolation algorithm, which may have some subtle issues.  It would be awesome if we could run the same file through marlin and see if there's a difference.  

It's still possible the welded code has some issues, but if that's the case I haven't figured it out yet.  If it is the welded gcode, it would have to be extrusion/retraction amount differences, since the actual paths look more or less perfect.  What version of ArcWelder are you using?

appleboblin, 2021-01-24:
I am using 1.0.0+u.bb71e8f, the one from OctoPrint plugin page. I will print the none processed side by side comparison once my current print is done. I have not yet tried printing A cylinder since upgraded to klipper, the one with arc welder was the first. Not sure if there will be noticeable seam or not for cylinders but other prints don't have seam that are very noticeable other than the first few prints I printed with cura.

FormerLurker, 2021-01-24:
OK, if the seam doesn't appear in the non-welded version, I may ask you to try the latest devel code just to see what happens.  Lots of changes have been made since the original release.

appleboblin, 2021-01-25:
[image]
I've just printed the non-welded version, but I think my printer currently have some extrusion issue. But I can defiantly feel the seam on the non-welded version more. I will post the result once I fix the extrusion issue

appleboblin, 2021-01-25:
Seems like I have more tuning to do when it comes to printing cylinder, both have the same issue. I think I'll have to play around with pressure advance a bit more. Thanks for your help!
[image]

FormerLurker, 2021-01-30:
Glad to be of assistance.  Since this appears to not be an arcwelder issue, I'm going to go ahead and close this out.  If you run into any additional problems, please let me know, and consider helping me test the newest release: v1.1.0rc1 (I can provide installation instructions if necessary).

Again, thanks for your post!

closed 2021-01-30 by FormerLurker