FormerLurker/ArcWelderPlugin #132: Set Color after processing

KenLucke, 2021-01-28:
It would be nice to have a quick and easy way to distinguish between a processed and unprocessed file without having to squint at the filename `<g>`

If there were an option to set a different color to a processed files, much as OP itself changes from black text to green after a successful print or red if it failed it would be a snap.  A optional color picker to choose your own color would be a plus to those using Themify and don't want to track down the CSS declaration.

FormerLurker, 2021-01-28:
What about using a prefix, like AW_?

KenLucke, 2021-01-28:
These old eyes still have to squint at the filename.  It was just a thought.

FormerLurker, 2021-01-30:
@KenLucke, what about using the 'Source File Deletion' option here:

[image]

Or, if you don't want the file name changed you could also use this option:

[image]

Finally, you can achieve the same effect by erasing the prefix/postfix:

[image]

Here's another idea I had:  What if a new plugin is created that can change the color of files based on a pattern or regex?  Maybe like ```*.aw.* = Red``` or something like that?  It would be much more powerful, and possibly work in many interesting situations.

The reason I'm providing these workarounds and alternatives is that ArcWelder already messes with the file manager a lot, and they are basically hacks (adding the buttons and arc welder info to the drop down).  If the file manager changes too much they may stop working.  I'd prefer to limit the the file manager changes to what I've already added if possible.