FormerLurker/ArcWelderPlugin #133: Issue with small cilinders

Delta-38, 2021-01-29:
Hi, I really like ArcWelder plugin, so much so I had turned it to automatically process all files I sent for print.
I'm printing on an Anet A8, for the most part I'd say it (AW) does a stellar job, but I have encountered a strange persistent issue with small cilindrical features. 
For a time I thought my motors must be going since I coudln't find anything different in the settings. 

The effect is that on tight cilinders (2mm -3mm) the nozzle appears to arc around the object with an offset on different layers, producing an effect similar to badly stacked circles. 
It does not appear to do this for larger cilinders. 

I tested with a set of 3 test objects, a small retraction test with two pillars, a thin square, and a single wider pillar. 

The square is identical in both tests, the single pillar is slightly narrower with AW enabled (3.45mm instead of 3.90 mm diameter without AW), but still good. 
The retraction test with the twin pillars instead shows severe issues (without AW pillars measure 1.95mm whereas with AW they go to nearly 3mm).

I am using Slic3r 1.3.0 on windows for slicing.

I am sending along a picture (with a batch printed without AW on the left, and the AW brethren on the right), the gcode and the stl for the offending pillars)
[repro.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5895401/repro.zip)

Thanks in advance,
Paolo Manili

FormerLurker, 2021-01-29:
See [this issue (enhancement)](https://github.com/FormerLurker/ArcWelderLib/issues/18) in the ArcWelderLib repo about this very problem.  The feature I linked to there is available in the devel branch if you want to try it out.  Let me know if you would like to help me test, and if you need instructions on how to install it.

I recommend setting min arc segments to at least 14, but no higher than 24.  The higher that number, the better the small curves will be, but the smaller the compression ratio.  For example, if you used too high of a number you may see 0 arcs produced.

The enhancement does exactly what it says, and you can get great results, but the REAL fix is to update your firmware to something that contains more modern arc generation settings and not encounter this problem in the first place.  That being said, I'm thrilled to have more testers for this feature since many printers are not easy to flash with different firmware!

Thanks!

Delta-38, 2021-02-02:
Hi, sorry for the late reply, it was a very busy couple of days. 
Thanks for suggesting a firmware upgrade, I didn't realize it could have that much weight on the results. 
I have repeated the test with the latest Marlin and it performed much better (though on stock AW settings there is still a bit of off center arching present).
The issue you linked explains the problem very clearly :)
For completeness's sake I am sending you the pictures and gcode from this latest attempt, with two from the stock firmware test. 
I would like to help with testing the development branch, how do I go about it?

PS. Stock Anet SA, Stock Arc Welder AW, Marlin M
Left to right: SA, SA-AW, M, M-AW 
[cilinder issue 2.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5909765/cilinder.issue.2.zip)

FormerLurker, 2021-02-02:
Hmm, looks like there is still a ways to go, if I am reading the snapshot correctly.

Thanks for helping with the new rc/maintenance build!  You can install from the 'Plugin Manger' with this URL (get more->From URL):

```
https://github.com/FormerLurker/ArcWelderPlugin/archive/1.1.0rc1.zip
```

I'm going to take a look at the gcode you posted later (might be the weekend) and see if there is any obvious issue.  Since the new version you will install has a very different algorithm at the core, you may want to try it out and see if your results differ.  I'm also interested in seeing what the firmware checker says, and what you think about the new UI in general.

Delta-38, 2021-02-03:
Hi, I'm going to try the development branch tonight (GMT +1), in the meantime I wanted to tell you about another curious behavior I noticed (running Marlin and AW released version) on a larger set of circular shapes, I was printing a Chess Queen and the result is quite unusual. 
I'm sharing the gcode and a picture as always (btw. sorry I stopped the print but I was quite afraid it might fail and had to leave the house):
[Queen code.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5916985/Queen.code.zip)

FormerLurker, 2021-02-03:
Something very strange is going on.  When you install the new version, look at the firmware checker.  I'm thinking you need to turn on the 'G90/G91 influences extruder' setting.