FormerLurker/ArcWelderPlugin #136: AW printing at sharp angles rather than curves OctoPi 1.5.3 / TH3D U1.R2.B5 firmware

Daveholmes05, 2021-02-03:
Hey all,

I've recently upgraded to an Ender 3 1.1.5 silent board and flashed firmware to TH3d U1.R2.B5 firmware on my printer. I noticed prints with curves were rough at the curved portion. 
[image]

I did some research and increased baudrate and buffer size and it helped but I am still getting rough curves. Someone suggested Arc Welder so I installed that plugin and tried it. Unfortunately, it's printing with a sharp angle rather than curving out like it should. The skirt shows the correct curvature but the actual print doesn't. I've uploaded a print (first time doing this, hope it attached correctly)
[image]

I tried changing the resolution in Arc Welder to 0.1 to see if it made a difference, and it did not. Also tried lowering it to 0.025. Same results.

I ran M115 to test my board and it doesn't say anywhere about arc support
Send: M115
Recv: FIRMWARE_NAME:Marlin TH3D U1.R2.B5 SOURCE_CODE_URL: PROTOCOL_VERSION:1.0 MACHINE_TYPE:TH3D U1.R2.B5 EXTRUDER_COUNT:1 UUID:
Recv: ok

I ran G2 and G3 and it says unknown command:
Send: G2
Recv: echo:Unknown command: "G2"
Recv: ok
Send: M105
Recv: ok T:150.20 /0.00 B:54.07 /0.00 @:0 B@:0
Send: G3
Recv: echo:Unknown command: "G3"
Recv: ok

My understanding is that TH3D firmware is just a form of Marlin, which I believe should be supported.

Can anyone provide some insight on how I can get AW to work for me? Or what I can do to get these curves to print better?

Daveholmes05, 2021-02-03:
Also forgot to mention I checked my firmware configuration_adv.h file for ARC SUPPORT and this is what it shown:
#if DISABLED(SLIM_1284P)
	#define ARC_SUPPORT             
#endif
#if ENABLED(ARC_SUPPORT)
  #define MM_PER_ARC_SEGMENT  1  
  #define N_ARC_CORRECTION   25  
#endif

seantapscott, 2021-02-03:
That's not an arc welder problem, unless you're expecting him to
troubleshoot your printer issues for you.

FormerLurker, 2021-02-03:
Thanks for posting your issue @Daveholmes05!  This is actually a well known issue on older versions of Marlin that don't have additional arc settings.  Your version is missing some critical settings for handling small curves, so you have two options to fix this:

1. Upgrade to at least Marlin 2.0.6, which has MUCH better arc support.
2. Upgrade the Arc Welder plugin to the latest maintenance release candidate and utilize a new feature called 'firmware compensation' that should deal with your issue at the cost of losing some compression around small arcs.  See [this issue in the ArcWelderLib](https://github.com/FormerLurker/ArcWelderLib/issues/18) repository (the heart and soul of the ArcWelder plugin) for more details about the problem and the workaround.

I do recommend option 1, but understand if you want to try option 2 (that's why I built it after all, lol).  If you want to give the maintenance release candidate version a go, you can install it via the plugin manager (get more->from url...) with the following link:

```
https://github.com/FormerLurker/ArcWelderPlugin/archive/1.1.0rc1.zip
```
Once it installs, enable firmware compensation here:

[image]

You MUST re-weld any file you want to print after enabling that option, because the setting will have no effect on previously welded files.

Also, if the quality of the small arcs is still not good enough, you can bump 'Minimum Arc Segments' up to 24 to achieve marlin 2.0.6 quality, but you will lose much more compression and may even get stuttering again.  I recommend trying it out with 14 and see how it goes from there.

Let me know how it goes, and if this advice was helpful at all!  Good luck!

Daveholmes05, 2021-02-03:
> Thanks for posting your issue @Daveholmes05! This is actually a well known issue on older versions of Marlin that don't have additional arc settings. Your version is missing some critical settings [...]

Thank you so much for the quick response. I am admittedly new to all of this. For some reason I thought I was on one of the newer versions of firmware. How did you identify I am on an older version? Is it from the protocol version 1.0 from my M115 command?

I don't remember how I determined it, but I seem to recall figuring out I was on an 8-bit board and that's why I chose the firmware that I did, but I could very well be wrong.

So can I just flash this Marlin firmware version and then I should be good? https://github.com/MarlinFirmware/Marlin/tree/2.0.7.2

Thanks!

FormerLurker, 2021-02-03:
@seantapscott, it's no problem since I need testers for the maintenance release candidate anyway, and I have all the info needed to diagnose this.  Thanks for patrolling, and those are some FANTASTIC Succulent Planters in your repo :)  I will have to print some.

FormerLurker, 2021-02-03:
> How did you identify I am on an older version? Is it from the protocol version 1.0 from my M115 command?

Unfortunately, the fork of marlin you are using has no indication of the base version it was forked from.  However, I recognize the arc configuration defines as being pre-marlin 2.0.

> I don't remember how I determined it, but I seem to recall figuring out I was on an 8-bit board and that's why I chose the firmware that I did, but I could very well be wrong.

Marlin certainly will run on an 8-bit board.  However, it's hard to tell if you'll have enough memory to load all the features you want without just trying.

> So can I just flash this Marlin firmware version and then I should be good? https://github.com/MarlinFirmware/Marlin/tree/2.0.7.2

Well..  You'll have to configure and compile it.  Not sure how much experience you have there so feel free to stop reading if you're an expert (I will assume you are not).  If I were you I'd hop on the Marlin discord server and ask a couple of questions.  It might be a good idea to start reading the docs and looking for pre-made settings files for your printer (they typically exist).  Maybe even try compiling the code and getting used to that.  The more you know, the less you need to ask, and the more willing people in the discord will be to answer your questions. I can tell you that compiling your own firmware and flashing it can be a challenge, but it is EXTREMELY satisfying, and gives you access to very interesting side of 3d printing that most people never attempt.  You will learn a lot from the process.  Once you've done it a few times you won't think it's such a difficult thing anymore :)

FormerLurker, 2021-02-03:
Oh, and if you're willing, I would love it if you could give the maintenance candidate a go and running a quick test to see if firmware compensation is working.  I desperately need more testers so I can push out the new code.

Daveholmes05, 2021-02-03:
Thank you very much for the additional info. I will try the maintenance candidate and see if that works. I'll let you know how it goes.

seantapscott, 2021-02-03:
@FormerLurker Yeah, sorry, been following what you're doing because I think it is awesome, and forgot that I'm trying to be a good person for a second. 

In other news, I just updated to one of TH3Ds Creality Ender 5 firmwares with Marlin 2.1.8, and the arcs are working great through the ArcWelder Cura plugin.

FormerLurker, 2021-02-03:
> been following what you're doing because I think it is awesome

@seantapscott, I'm touched, thanks!

> and forgot that I'm trying to be a good person for a second.

lol!  I wouldn't worry too much about it.  The open source community is constantly being bombarded on all sides with every negative emotion imaginable, and it makes it very difficult to figure out what should be done about it.  In fact, I recently was invited to a sort of support group for open source devs to try to get a handle on this.  It's a constant struggle, and I have my good and bad days.  The important thing is you are contributing, and that is admirable!

> In other news, I just updated to one of TH3Ds Creality Ender 5 firmwares with Marlin 2.1.8, and the arcs are working great through the ArcWelder Cura plugin, so it sounds like you've nailed it.

Are you using the latest development snapshot?  I'm quite interested in seeing how that is going, since it uses the same arc generation engine that's in the latest OctoPrint plugin release candidate.  It's a huge upgrade, and is very difficult to test given the sheer volume of slicers and firmware out there.  I've yet to find provably 'bad' gcode output, but I've seen SO SO many edge cases and unexpected gcode that I'm willing to be there are some bugs lurking in there somewhere.

Anyway, keep on keeping on, and I hope that you are enjoying what you are seeing in this humble collection of repos!

Daveholmes05, 2021-02-03:
Unfortunately, the maintenance candidate made things worse. It's still printing the sharp angle rather than the curve but now it's pausing at two spots and causing a big blob. It caused the first test (default 14 minimum arc segments) to fail and get ripped off the bed plate. I was unable to cancel the print after numerous tries (normally I can cancel the print and it stops after a couple of seconds) and had to flip the power switch on the printer to get it to stop.
[image]

Second attempt I changed the minimum arc segments to 24 and generated a new file from cura (to ensure it was all a fresh file conversion and not just a re-convert). Same exact scenario. Pausing, blobbing, failed print, couldn't stop:
[image]

FormerLurker, 2021-02-03:
That's unfortunate, but thanks for trying!  Can you send me the before and after gcode please?  I'd like to make sure the gcode itself is correct, and that it's not some kind of undiscovered bug.  Please use gist.github.com to upload the files, and drop a link in here, thanks!

Daveholmes05, 2021-02-04:
Sorry, what's the best way to get that to you? Where do I collect what you are wanting?

FormerLurker, 2021-02-04:
You can turn on serial logging in the octoprint settings under the log menu.  Just be sure to turn it off when you are done!  Then print a layer and download the serial log file from the log menu.

Daveholmes05, 2021-02-04:
I think I did this correctly. I created a new file from Cura that was only about 4 layers tall to have a quick beginning and end print for you. Hopefully it was a good move.

What Cura shows in preview:
[image]

What it printed (note that the skirt was only printed vertically on the left side of the print)
[image]

Here's the gist link:
https://gist.github.com/Daveholmes05/b00e1344506d6108a2228303d24a4d5a

FormerLurker, 2021-02-04:
Bingo: ```echo:Unknown command: "G3 X91.100 Y131.958 I7.214 J-0.941 E23.39124"```

Arcs are disabled

FormerLurker, 2021-02-04:
I am surprised the firmware checker didn't report that.. hmm

Daveholmes05, 2021-02-04:
So is there a way for me to enable arcs for this firmware or do I just have to try and flash to a newer version?

FormerLurker, 2021-02-04:
You would need to recompile it.  Might as well use the newest version, right?

Daveholmes05, 2021-02-04:
OK, so I should just try this latest version then? https://github.com/MarlinFirmware/Marlin/tree/2.0.7.2

Do I need to check anything to make sure arcs are enabled when I flash it to a newer version?

FormerLurker, 2021-02-04:
Yes, newest version is always the best bet.  I believe arcs are enabled by default, but just in case, make sure this section in Configuration_adv.h looks  like this:
```
//
// G2/G3 Arc Support
//
#define ARC_SUPPORT                 // Disable this feature to save ~3226 bytes
#if ENABLED(ARC_SUPPORT)
  #define MM_PER_ARC_SEGMENT      1 // (mm) Length (or minimum length) of each arc segment
  //#define ARC_SEGMENTS_PER_R    1 // Max segment length, MM_PER = Min
  #define MIN_ARC_SEGMENTS       24 // Minimum number of segments in a complete circle
  //#define ARC_SEGMENTS_PER_SEC 50 // Use feedrate to choose segment length (with MM_PER_ARC_SEGMENT as the minimum)
  #define N_ARC_CORRECTION       25 // Number of interpolated segments between corrections
  //#define ARC_P_CIRCLES           // Enable the 'P' parameter to specify complete circles
  //#define CNC_WORKSPACE_PLANES    // Allow G2/G3 to operate in XY, ZX, or YZ planes
  //#define SF_ARC_FIX              // Enable only if using SkeinForge with "Arc Point" fillet procedure
#endif
```

There are other configurations that will work, but these defaults are quite good, and should produce very nice arcs.

seantapscott, 2021-02-04:
Should look on Th3dstudios.com website. Their latest versions support arcs
on creality printers quite well.

Daveholmes05, 2021-02-04:
> Should look on Th3dstudios.com website. Their latest versions support arcs on creality printers quite well.

Yeah, for some reason I thought I could only use the 8-bit version because I believe my board is 8-bit. So I am pretty sure I downloaded the latest 8-bit version but I could be wrong. I will check again.

seantapscott, 2021-02-04:
I think their instructions encourage us to compile it ourselves, so might
be able to just enable the arc support setting in the advanced
configuration file like FormerLurker stated.  That's probably the best bet.

FormerLurker, 2021-02-06:
@seantapscott, It may be to save memory, or just because some people are in the habit of disabling arcs.  After all, they were very rarely used until very recently.