FormerLurker/ArcWelderPlugin #137: How can I improve this cylinder print?

paulharman, 2021-02-04:
First off, thanks for your work on this :)

Here I have an example print, PETG, printed at 235/80 with no fan, 50mm/s. I'm using Cura, Octoprint and arc welder with the CR6 community fw on a BTT SKR CR6 board . The model is hollow with a 1.8mm thickness. The G-code file is 1.74 MB after arc welder (15.3MB before).

If I print from SD card the print is fine without the zits.  Can you recommend any settings which may help to improve it?

[hoover.aw.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5922599/hoover.aw.gcode.zip)

[image]

[image]

seantapscott, 2021-02-04:
These look like connection issues.  Is it possible to change the input
buffer length?  In RepetierServer, I print over the USB, and I do
connection testing to find the best length for the input buffer.  If the
command runs too long, it timeouts as well, and the printer and the gcode
server (octopi in your case) will have a lot of chatter back and forth to
figure out which line they're on.

Are there any communication troubleshooting tools in Octopi?

FormerLurker, 2021-02-04:
Maybe create a serial log on octoprint and check for numerous resend errors?  A couple layers should be enough.  Also, try rebooting into safe mode and run another test.  That looks pretty excessive.  Maybe also run a few layers from sd and see if things change.