FormerLurker/ArcWelderPlugin #138: And now for something completely different...

KenLucke, 2021-02-04:
"It's....." [Sousa's "Liberty Bell" (Monty Pyton Theme) plays...]

Uploaded a file while AW was set to Manual Process Only.  Realized it WAS set to Manual Process Only, so changed settings to "Process All New Files".  Clicked "Weld Arcs" icon in file menu.  AW said it was starting to process the file, then claimed it could not find the file... even though it was clearly visible in the file browser.

100% reproducible here:

1. Turn off Process All New Files [set to Manual Processing Only]
2. Upload a file (I used the upload button, probably doesn't matter if it were drag & dropped)
3. Turn ON Process All New Files
4. Attempt to Weld Arcs on just uploaded file.
5. Fail

[image]

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:85.0) Gecko/20100101 Firefox/85.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 8140980224
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.4
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 21.0.1
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin 2.0.7.7B_NIC (Dec 31 2020

FormerLurker, 2021-02-04:
Awesome instructions and neat info below!  Will check this out for sure.

FormerLurker, 2021-02-04:
Actually, looks like it may be URL encoded, but not decoded properly.  If so it should be an easy fix.

KenLucke, 2021-02-04:
> Actually, looks like it may be URL encoded, but not decoded properly. If so it should be an easy fix.

Don't you know by now that nothing I bring up is EASY????

FormerLurker, 2021-02-06:
OK, it appears to have something to do with the file name (again, probably Url encoding).  You have the craziest file naming scheme I've ever seen, so it's not too surprising, lol!  I'm going to ATTEMPT to hand type one of these in, but if you could just copy and paste in the file name you are using, that will probably save my skin.  I'm going to basically just us a name with the characters I suspect are causing the issue, but I'd like to verify the names you are using don't have some sinister character lurking in there that I can't see.

KenLucke, 2021-02-06:
I use the Cura extension that adds many of the slicing parameters to the filename [Gcode Filename Format extension].  I have several printers, and often have different nozzle sizes installed across them, as well as other differences.  I also often slice the same file for different materials, or am trying different layer height, speed, infill etc, tweaks.  The inclusion of that information in the filename gives me a way to tell, at a glance, what version of a file I am looking at.

Here's that file's filename as it exists on my Mac:
`Puzzle Box [PETG] [skirt]+Infill [20%,cubic]+Noz [0.6]+LH [0.2]+LW-[0.6]+Temps [255+70]+75.0mms+Support [normal (56)]+Coast-[True]`

That's the result of the following settings in the Gcode Filename Format extension in Cura:
`[base_name] [[material]] [[adhesion_type]]+Infill [[infill_sparse_density]%,[infill_pattern]]+Noz [[machine_nozzle_size]]+LH [[layer_height]]+LW-[[line_width]]+Temps [[material_print_temperature]+[material_bed_temperature]]+[speed_print]mm/s+Support [[support_structure] ([support_angle])]+Coast-[[coasting_enable]]`

FormerLurker, 2021-02-06:
Thanks!  I think I've solved the original issue, but now I have to deal with the display name as well.  Arc welder isn't currently smart enough to handle both display names and slugified real file names.  Many of the characters in your original file are replaced, and AW is showing the slugified name, which should not be altered.  Sigh....  LOL!

FormerLurker, 2021-02-06:
Ok, you know the drill.  Install with this URL and test the $hi#@@# out of it :)

```https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip```

It should not only process now, but display the correct name in both the file manager, the arc welder popup notifications (please enable those if you have them disabled for testing), and the statistics window.  If you check in the OctoPrint data folder, you should see a slightly different slugified name (percent replaced with dash, some other small modifications).  Arc welder now tracks source file display name, source file disk name and the the target file display name and disk name :)

KenLucke, 2021-02-06:
Hey, I toldja.  my issues are never simple.  :)  When you push software to its limits, you find breaking points.  Which is why I've alpha/beta tested since the early 80s.

KenLucke, 2021-02-06:
> Ok, you know the drill. Install with this URL and test the $hi#@@# out of it :)

Fortunately, some of my printers are between print jobs right now, so I can test.

KenLucke, 2021-02-06:
I just noticed that I had not sponsored you.  I just did.   I really thought I had done that long ago.  My apologies for being so thoughtless, you've always been incredibly responsive and have treated me as if I WAS a high level sponsor.

KenLucke, 2021-02-06:
> > Ok, you know the drill. Install with this URL and test the $hi#@@# out of it :)
> 
> Fortunately, some of my printers are between print jobs right now, so I can test.

all appears to be good.  Used the same raw file off of my computer, both via drag & drop and upload button, followed the same reproduction steps I listed, and everything looks perfect.

OOOPS,  no, I lied.  Everything with AW looks perfect, but the "internal filename" display in the file browser (which I hate, believe serves no useful purpose, asked for the ability to turn off [which they added] and now consequently always leave disabled [except for this test]) is gone after the first AW processing, and will not return regardless of toggling the setting in the Appearance panel, even after a reload or even a restart.  Did you throw a switch somewhere in your code you forgot to turn back on?

[image]

FormerLurker, 2021-02-06:
I was unaware of that option.  However, FWIW, here's what I see in the sidebar:

[image]

And when unchecking that option:

[image]

Seems like an upstream issue, no?

FormerLurker, 2021-02-06:
Oh, I just realized, it looks like there are no percents in the file names in your example.  In that case there would be no internal fine name display, since the source and target file names would match, so I don't think this is a problem (unless I am mistaken).

KenLucke, 2021-02-06:
In your example, you're using the raw formatting string from the Cura plugin, which is not an actual filename... don't use that, use the actual filename.

As for the % signs, they're indicating a character code preceeds (20% means a space, etc.)

KenLucke, 2021-02-06:
> Oh, I just realized, it looks like there are no percents in the file names in your example. In that case there would be no internal fine name display, since the source and target file names would match, so I don't think this is a problem (unless I am mistaken).

It's always shown it regardless, as long as there was a single character that Linux had to translate.  That's why I requested (and got) the option to turn that garbage I don't need to see off :).

FormerLurker, 2021-02-06:
Which character did you expect to be translated?  I will run this and see how the path on disk compares to the display path

FormerLurker, 2021-02-06:
Hmm. Was there a % in the original? It is hard to see from your snapshot

KenLucke, 2021-02-07:
> Hmm. Was there a % in the original? It is hard to see from your snapshot

> Here's that file's filename as it exists on my Mac:
Puzzle Box [PETG] [skirt]+Infill [20%,cubic]+Noz [0.6]+LH [0.2]+LW-[0.6]+Temps [255+70]+75.0mms+Support [normal (56)]+Coast-[True]

KenLucke, 2021-02-07:
> Which character did you expect to be translated? I will run this and see how the path on disk compares to the display path

We're way off of the original issue... I didn't have a problem with any of the translations, except in the case where it failed after switching auto/manual mode.

FormerLurker, 2021-02-07:
> You apparently grabbed the format string from Cura instead of the resulting file name.

Whoops, that's what I did.  Whoops.

> We're way off of the original issue... I didn't have a problem with any of the translations, except in the case where it failed after switching auto/manual mode.

Ok, so that one is solved.  I will try the original file name and make sure it works correctly before I close this out, unless I missed some other text.  I'm getting confused more than usual today.  Been going through lots of issues.

KenLucke, 2021-02-07:
> Whoops, that's what I did. Whoops.
>
>
> Ok, so that one is solved. I will try the original file name and make sure it works correctly before I close this out, unless I missed some other text. I'm [...]

The original filename works fine, as reported back a ways.

Relax, have a beer, forget your issues for a while :).

KenLucke, 2021-02-07:
Shoot me a PM with your PayPal address, please :)

FormerLurker, 2021-02-07:
Even easier:

[image]

😄

KenLucke, 2021-02-07:
Ah, missed that.  Not everyone has it configured.

closed 2021-02-07 by FormerLurker