FormerLurker/ArcWelderPlugin #139: Problem with encoding characters

ahernsean, 2021-02-04:
When I use the Arc Welder plugin in Cura 4.8, I can never save the GCODE file. The only difference in the slicing settings is whether I have the Arc Welder checkbox on under Special Modes. The error I get when I save the GCODE is:

`'charmap' codec can't encode character '\ufeff' in position 757: character maps to <undefined>`

[image]

I've gotten this no matter what I slice. Turning Arc Welder off allows me to save the GCODE, but (of course) prevents Arc Welder from running.

FormerLurker, 2021-02-04:
This is an issue with the cura plugin that has been fixed in the latest development snapshot.  I don't maintain that plugin, but I know there is a vetting process before these versions can be installed via the interface.  You could try out the development snapshot, or wait a week or two until that build is available from the marketplace.

ahernsean, 2021-02-04:
That's great news. Sorry to clutter the Issues here with someone else's problem. Is it a development snapshot of Cura itself?

closed 2021-02-04 by ahernsean

FormerLurker, 2021-02-04:
No, it's a snapshot of the cura plugin itself.  I'm not 100% sure where to get this, but if you can't find it, message me back and I'll do a quick search.  I have it around here somewhere, lol

ahernsean, 2021-02-04:
Thanks. This is great news. I'll go figure out where to find that development snapshot.

ahernsean, 2021-02-04:
Hm. The best source I have is here: https://plugins.octoprint.org/plugins/arc_welder/, which ultimately points me here to this GitHub project. I don't see development versions listed.

FormerLurker, 2021-02-04:
I think [this is it](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ArcWelderPlugin/zzHJfdU_udY4lWeMO9r9Ha-zSXKtZbLMRyRejKldEXtt/ArcWelderPlugin-v7.0.0-2021-01-25T11%3A08%3A03Z.curapackage?Expires=1926932894&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=N8jg1rq3JW0q%2BeRHxmIRNaq5Lb87wH0yEoUrNUB6jLMAkVWlnKGGSHsnnqWhVMa7VYikI6vDzdxju3JGl0QS4CUI8h3HCRoWdlrmaYBb6oU9rm2E8pknuTxHWEmNdTjEyAJ3FhKANYi2%2BoYE8NsoxAwA%2FLmXEKdr%2FGE8smqWV4qgWhJNv1eITT%2F8pKChkZR3MmFIF%2FzjHHjSLWks4UjLeX03IUYTrozi3tpv2E80ebfXDzEtQMvTALsVjoDSOByy7xWNId3zAC2iZAdxplzyW54z8pW3xOYLae3lM0cHLhcfbOJMBGLctntAehlXes0DmhLvEjptwbPCRvBylaHCGg%3D%3D), but don't quote me :)