FormerLurker/ArcWelderPlugin #142: Autostart after pre-processing

NikopolCZ, 2021-02-05:
Hi, this is not really an issue but a nuisance. 

I am using script in s3d, so my prints start printing right after upload to octoprint. But this feature is not compatible with AW. Would it be possible to add an "Autostart after processing" function? It would save me a lot of time. 

Btw thanks for the pluggin. It has solved my problem with complex prints crashing.

swheettaos, 2021-02-05:
This is also impacting Cure "send to printer" option when you select "print after upload".  Also I use the astroprint plug to print from the cloud sometimes.  Same problem but I haven't found a work around.  It's almost like we need a buffer .. like if arc weld enabled, it grabs it .. and processes it.. then will allow it to start printing.

FormerLurker, 2021-02-05:
This feature exists on the beta.  Wanna try it out?

Also, for cura, consider using the cura arcwelder plugin, as it plays nicely with the cura octoprint plugin.

swheettaos, 2021-02-05:
I Will do that.. just switch to dev level updates right?  Also for using the Cura plugin, I can try it but I'd rather have one place where I'm dealing with settings and updates to configure arc welder.  It also allows me to be able to print from multiple slicers, (cura, prusa, cloud based) and all take advantage of the arc welder plugin.  I'll let you know.  I have about 7 hours on a print left and then I'll update to the beta and post results.

FormerLurker, 2021-02-05:
You can integrate arcwelder.exe into all slic3r varients and simplify, fyi.

Also, i goofed on the auto updates for v1.0.0.  i will fix it with a PR in OctoPrint itself hopefully.  For now, check the discussions tab in this repo and see the stickied post.

NikopolCZ, 2021-02-05:
I have tried the beta release and it is exactly what i wanted. I am going to close this issue. Thank you!

closed 2021-02-05 by NikopolCZ

swheettaos, 2021-02-05:
So I tried the beta release. In Cura I have to unselect the "print after upload" and in arc weld, I select print after processing.  Is that how you wanted me to do it? I haven't tired with Astroprint, there isn't an option to upload only.  Should I keep the setting print after upload in Cura?  If so what setting should I have in the Arc Weld plugin.  The other part is there are other things that connect to cura and upload and then tell it to print, how will this be affected?

FormerLurker, 2021-02-05:
You have two options:

1. In the Cura Octoprint plugin, disable 'Print after upload' and enable "Print after Preprocessing" in the ArcWelder plugin for Octoprint.
2. Install the 'Cura Arcwelder Plugin' and use the Cura Octoprint Plugin with the 'Print After Upload Option' and forget about Arc Welder in Octoprint.

If you try to use the 'Print After Upload' option in the Cura Octoprint plugin AND print after preprocessing option in the ArcWelder Octoprint Plugin settings, it will be a race to see which one gets the print started first (spoiler:  it will be the Cura Octoprint plugin 99% of the time)

swheettaos, 2021-02-05:
So I'm doing the first which works for Cura. The problem is I also use Astroprint and there are other plugins out there for cloud based stuff that then sends the print to octoprint like simplyprint.io, astoprint, raisecloud .. and other plugins like cura, prusa etc that connect to octoprint, upload a gcode file and then send the print command right after.  Most of these don't have and option to only upload, nor do they have a way to incorporate arc weld into their post processing.  I'm trying to find the best way to do this with other slicers and other cloud based solutions.  Suggestions?

FormerLurker, 2021-02-05:
Ok, so use arcwelder for cura, and install arcwelder for octoprint.  Arcwelder will not reprocess previously welded files, so it wont print any welded files sent from cura if you enable the 'print after processing' optioy.  Will that work?

swheettaos, 2021-02-05:
huh?  Lets take cura out of the picture, if we talk about any other slicer that connects to octoprint, they expect to upload and then start printing right away but I want Arc Welder to auto-process the files beforehand for any slicer that uploads.  Right now if a slicer uploads it say print right away and there there is a conflict with overwriting.. meaning it will process but then errors saying it can't overwrite because the file is being used for printing.  Did that change?  I only tested from cura so far but I think I will have the same problem that I did before the beta version.  The beta works for cura because I can tell arc weld in octoprint to auto-print,, and in cura upload only.. all the other slicers, upload and try to print right away.. which doesn't allow for arc welder to get in between the upload and processing.. or does it and I'm just talking out the wrong hole?

FormerLurker, 2021-02-05:
Sorry, I've been answering a ton of questions today, so I hope I'm getting what you are saying.  Let me simplify and rephrase.

You want to weld files within your slicers and then upload and print, but most slicers are totally unaware of anything that happens in Octoprint (and Octoprint is pretty much unaware of what the slicer is doing).  So, if you tell them to 'upload and print' (some not having the option to simply upload, but only upload and print), you get an error message.  Do I have it correct?

So, what we basically have here is a race condition.  Two competing processes are trying to be the first to start a print (the slicer and the OctoPrint ArcWelder plugin).  The one that 'wins' gets to start the print, and the one that 'loses' will trigger an error.  It's hard to predict which one will win this race, hence the 'race condition'.

Since you have some slicers that don't have the (somewhat reasonable) option of uploading without printing, you will need to integrate ArcWelder with those slicers.  If you let me know which slicers besides cura you want to use, I can tell you how to get the console application to post-process the gcode files.  I'm assuming (perhaps too much) that any slicer that offers post-processing capability will post process THEN upload and print.  If that's the case, you can get everything you want I believe.

You already know how to get welded files out of Cura:  Install the Cura Arcwelder Plugin and the Cura Octoprint plugin, and weld inside of cura.  Then tell the Cura OctoPrint plugin to upload and print.  That one is solved.

You can also integrate ArcWelder with PrusaSlicer/Slic3r/Super Slicer by using Print Settings->Output Options->Post processing scripts by using the following template:  ```{PathToArcWelderConsole} {ArcWelderOptions}```

For example:  ```C:\ArcWelder.exe```

You may need to enable 'Advanced' or 'Expert' mode.  It may be slightly different in Slic3r/Super Slicer or various other forks.  I have only tried this with Prusa Slicer, but have been told Super Slicer works too.

Or in Simplify3D you can edit your process, go to the 'Scripts' tab, and use the following template:

```{PathToArcWelder} {ArcWelderOptions} "[output_filepath]"```

Note:  The quotes are ESSENTIAL, else it may fail depending on the output folder and file name.  Here is an example:

```c:\ArcWelder.exe -r=0.025 "[output_filepath]"```

Just be sure if you are using Marlin 2.0, Prusa Buddy (for the mini, not MK2 or MK3), or smoothieware that you use the -g option.  If you are using Marlin 1 or a fork of Marlin 1 (or any older misbehaving firmware that has trouble with arcs of a small radius, like between 1 and 5mm) that you add the following options:  -s=1.0 -r=14

The s and r parameters add a feature called firmware compensation that fixes the small arc problem at the cost of a bit of compression.  If you don't need it, you don't want to use that.

Ok, got a bit sidetracked.  Now we need to deal with other apps that upload, but do not weld or start a print.  For that, install the ArcWelder plugin for Octolapse, enable the print after processing.  The ArcWelder Octoprint plugin will know if a file has been welded or not (it scans the file for the ArcWelder header).  If it has already been welded, it will ignore the file, so this will work just fine with your slicers that have already welded your gcode.  It will also work fine if you drop the file into OctoPrint (will weld and then print).

Did I cover everything?  Is there a scenario I missed?  Am I understanding you or am I way off?  Let me know.

FormerLurker, 2021-02-06:
OH, in case you don't know how to get the arcwelder binaries, see [this discussion](https://github.com/FormerLurker/ArcWelderLib/discussions/38).

swheettaos, 2021-02-06:
You covered everything for slicers I'm running on my machine and thank you for that I could make that work if that's all I was using.  I also use cloud slicers built into online solutions like simplyprint.io, astoprint, makeros etc.  While they have the ability to create or modify to some extent the slicer profile in there cloud slicers, they don't have a post-processing section where I could use the arc weld executable.  I REALLLY like arc weld and want to use it on anything that gets uploaded to octoprint no matter the method before I print but like you said there is a race condition.  Could we make it that if auto print after process is enabled in arc-weld that ignores the "start print" command from anything else? Or maybe have this as an option and just returns a success or something?  That way I can inject arc weld at the octoprint level between the upload and print process?

FormerLurker, 2021-02-06:
So your cloud slicers have no ability to just upload?  That sounds like something they should add...

Unfortunately they are probably sending a 'select file' and 'start print' command blind.  Arc Welder has no way, as far as I know, to tell who sent this command (could be from the OctoPrint web page, a cloud slicer, or cura).  I spent days trying to get this working for the Cura Octoprint plugin, and eventually the creator of that plugin just decided to make the cura arcwelder plugin himself, since it seemed the simplest way.

In short, I think you should ask the maintainers of these cloud slicers and ask them why no simple upload command?  Seems like a reasonable thing.  I often slice and upload several files at once, but don't print any/all of them right away.

FormerLurker, 2021-02-06:
What cloud slicers have this problem, if I may ask?

swheettaos, 2021-02-06:
as far as I can tell .. none of them have an upload-only feature ..  I will reach out to them.

FormerLurker, 2021-02-06:
Ok, sorry about that :(

Dr-Ari-Gami, 2021-09-19:
I know this issue was already closed, and that there is something in the beta (I haven't tried it), however, I have an idea. Please ignore me if this was already implemented.

Is there a way to use a print starting in octoprint as a trigger to arc weld the current file? 

Octolapse has a way of inserting gcode commands after the print has started, shouldn't this be possible for arc welder?