FormerLurker/ArcWelderPlugin #143: On startup Cura 4.6.1: ArcWelder.exe has stopped working

DenDuze, 2021-02-06:
Hi,

I just installed Arc Welder in Cura 4.6.1 and when I now start Cura I get the error "ArcWelder.exe has stopped working".
Already uninstalled Arc Welder and then Cura starts normal.
After new install of Arc Welder Cura I again get that error.

Is this a known issue? How to solve?
What can I do to debug the problem? Logfiles?

Regards

FormerLurker, 2021-02-06:
you using a mac?  If so, check out the [latest arc welder binaries](https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.1.0), which finally contains a working mac binary (thanks @fieldofview!)

You'll have to find the ArcWelder.exe file (buried somewhere in your cura folder I suspect, so do a search) and replace the exe.  Let me know how it goes!

DenDuze, 2021-02-07:
No I'm not using a Mac
I'm using Windows 8.1 Pro
Isn't there some possibility to enable logging so maybe we can find what teh problem is.
I also installed Arc Welder on my Rpi with Octoprint so I can use it but I guess its' better do do this from Cura so my gcode files are smaller when i copy them to my Rpi. So would like to run it from Cura.

DenDuze, 2021-02-07:
Today I saw that there was an update available in Cura for Arc welder ...Updated and now cura starts without problems and i see the Arc welder in the Special Modes so problem is solved.
Thanks

closed 2021-02-07 by DenDuze

FormerLurker, 2021-02-07:
Awesome, thanks for reporting back @DenDuze!  I suspect your problem was because older versions of the windows app required the windows C++ runtimes.  Those are now statically linked.  I'm super glad it's working for you!  Enjoy the plugin!

fieldOfView, 2021-02-07:
For future reference, the issue queue you posted in is for the OctoPrint plugin. The Cura plugin has its own repository (and its own maintainer):
https://github.com/fieldofview/Cura-ArcWelderPlugin

FormerLurker, 2021-02-07:
@fieldOfView, thanks!  I added a new 'Wrong Repo' label, and will start directing ppl there.

Also, I'm about to drop a new set of binaries.  I found a substantial bug triggered by line breaks between gcodes that are translated into an arc.  It was actually two bugs conspiring together, and can result in out-of-order gcode.  I believe cura is immune, except within the start gcode (where arcs should not be generated anyway) and layer change scripts (where arcs should not be generated anyway).  I'm finishing up the readme.md, and will post new binaries ASAP.  I'm also still trying to get hold of a MacOs VM.  Will keep you posted.