FormerLurker/ArcWelderPlugin #149: Cohesion3d LightBurn Laser Cutter Implementation

B-Man1, 2021-02-08:
Can Arc Welder be used on any g-code file?  I have a laser cutter that uses the Cohesion3D laserbaord which uses Lightburn software and g-code to run.  I also have a CNC router that can use g-code.  Both would benifit from ARC welder in the same way printers do.  Is it as simple as running the G-code file through ARC Welder then putting it on the SD card for the laser and running it?  It would be sweet if ARC Welder was integrated into Lightburn software so that this step could be skipped

FormerLurker, 2021-02-08:
Hi!  Can you send me a sample gcode file for both?  I have a feeling ArcWelder will not generate any arcs, since it does not encode travel moves ATM (it was designed for 3d printers), but I might be able to make it work.  I've never had a sample before to work with.

B-Man1, 2021-02-08:
I have attached a circle as a random test, see screenshot below and g-code
attached

[image: image.png]

This is for a laser cutter so only 2d and 2-axis.  The Cohesion3d
Laserboard does run Smoothieware however and can be used in printers etc so
maybe it will just work.  I'm curious to see what you find out

Thanks a lot

Brian

FormerLurker, 2021-02-08:
Hmm, your code didn't post.  Why don't you drop the gcode into gist.github.com and leave a link here?  Adding it to the email and replying doesn't work.

B-Man1, 2021-02-08:
https://gist.github.com/B-Man1/f0437cf4a4792a998e8b5ea404b29177

FormerLurker, 2021-02-08:
That worked.  Looks like I just need to add support for encoding travel moves too, which is what I had hoped.  It won't be a walk in the park, but shouldn't be too bad.