FormerLurker/ArcWelderPlugin #150: Catastrophic error when printing

Troutzorz, 2021-02-08:
Cura 4.8, ArcWelder Cura plugin 3.5.1 
Printer FLSUN QQ-S Pro
I tried ArcWelder for the first time, using the cura plugin. The print was going great until a little over 23mm Z when the effector swung wildly off course and crashed into the tower. When I tried to stop the print, it seemed frozen so I had to cut the power to recover. If I was not there to cut the power, this could have badly damaged the printer.
I am including the gcode file which caused the error.
[Roller Handle.gcode.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5947106/Roller.Handle.gcode.txt)
And an image of the incident.
[image]

FormerLurker, 2021-02-08:
Hmm.. sorry you are having issues.  I think you're the first person I've seen who is using Repetier, so maybe this is connected somehow.  

I ran your code through two different visualizers to see if I could figure out what's going on, but didn't see anything obvious, which doesn't mean this is not a problem with ArcWelder itself.  Here is simplify3D's visualization:

[image]

And the PrettyGcode viewer (it uses marlin's arc interpolation algorithm, so it is very accurate):

[image]

Can you see if you can pinpoint exactly where on your print the issue happened?  I see that little string, but I'm wondering if we can tie it to a specific piece of gcode somehow.  Also, if you have calipers, can you measure the exact height?  I have to go through this gcode by hand to figure it out, and narrowing it down will save hours of work.

Next, would you be willing to try installing the latest version of the console application and run your original gcode through that and send me the results (or just post the original and I can do it).  I did not write, nor do I maintain the cura plugin, so its simpler for me if you use the console version.  I added a bugfix for one issue that could cause problems (however, I would expect the issue to be obvious, but maybe not).  

If I can find changes, or pinpoint the gcode, we might be able to run a (highly) supervised test on your printer.  I'd like to know if the gcode is bad, but looks good, or if the firmware has some issues running arcs (I can add those to the firmware tester).

Thanks for posting!

Troutzorz, 2021-02-08:
[image]
[image]
[image]
Here are the images, I circled the area where it looks like it went off the rails.
I don't have the original gcode, I think the plugin overwrites it. 
I have a hunch that a tiny radius arc might cause the firmware to misbehave

FormerLurker, 2021-02-08:
Awesome photos, thanks!  I should be able to narrow it down quite a bit from this.  I will also look at the firmware source and add it to the inverse processor.  That should give us a lot more info.

Troutzorz, 2021-02-08:
The control board is an MKS Robin Mini. I think it is a custom firmware developed by the manufacturer, and sadly closed source.
I attached the .bin file of the firmware (with .txt extension so github would allow it).
[robin_mini.bin.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5947542/robin_mini.bin.txt)

FormerLurker, 2021-02-08:
@Troutzorz, the bin is not particularly useful, but thanks for posting it!  I might be able to glean some info from it.  The thought of decompiling it or working out the assembly makes me shudder, lol!  Still, if it is a fork of Repetier (I have some connections into that world), I might be able to figure out what is going on.  However, I don't want to get ahead of myself.  Step one is to make figure out of there is some obvious gcode issue at or near the point of failure.  If the gcode is accurate and valid, the next step is to figure out why the firmware had problems.  If the gcode is broken, I will be able to fix the issue myself.

This is no small task, but I will take it one step at a time and will do whatever I can to fix any issues that I discover.

kevmoso, 2021-02-09:
I have the same printer having the same issue with ArcWelder.
I however cannot provide such useful measurement information as I threw all the failed prints away.

FormerLurker, 2021-02-09:
Can one of you send an M115 command to your printer and paste in the results?  I can update my firmware Library at least and warn others that way hopefully.

susisstrolch, 2021-02-09:
Because I got mine QQ-S last week and because I'm curious like hell I tried to print @Troutzorz gcode, running a telnet monitor to the printer.  Error at 23.35mm...

Maybe it helps a bit to find the reason...
Unit seems to be in 0.01mm
[issue150.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5950386/issue150.txt)

FormerLurker, 2021-02-09:
Ok, I have it down to an exact layer now, and i was able to find that error in the Repiter github:

https://github.com/repetier/Repetier-Firmware/blob/2bbda51eb6407faf29a09987fd635c86818d32db/src/ArduinoDUE/Repetier/motion.cpp#L2145-L2148

Not sure what is going on yet, but it looks like i have yet another firmware interpolation routine to integrate into ArcStraightener to figure this out :)

We can ALMOST safely label this a firmware issue now, but I'm going to try to run this on my hardware if possible first, to make sure it prints well on at least one printer.  If it fails, then it's a gcode problem.

kevmoso, 2021-02-13:
Hope this is helpful.

Send: M115
Recv: ok 2
Recv: FIRMWARE_NAME:Robin
Recv: Printed filament:34890.38m Printing time:10 days 12 hours 42 min
Recv: PrinterMode:FFF

FormerLurker, 2021-02-22:
Quick update:  I'm doing some work on ArcStraightener, and am planning to add two new interpolation routines, one of which will be Repiter.  I will also add Klipper, and potentially some marlin forks.  I'm hoping this will lead to a better understanding as to why you are seeing these failures.

ColinBathe, 2021-04-23:
@Troutzorz this may seem a strange question but are you printing from USB or SD card? If USB, does the same problem occur if you print from SD card?

It may be completely unrelated but I'm having similar (ish) issues with my Sidewinder. Not got the bottom of it yet. I need to try out the latest bugfix branch when I get a moment.
https://github.com/fieldOfView/Cura-ArcWelderPlugin/issues/15#issuecomment-812504580

chromoxdor, 2021-05-07:
Just wanted to add me to the list with my flsun q5 (very similar to the qq-s).
Happened three times now. one time after 25 layers it went down suddenly down and scratched over the glassbed.

> Send: N4039 G2 X-1.302 Y-1.229 I3.488 J52.549 E511.67337 F3000*10
> Recv: ok 4039
> Send: N4040 G1 X-1.307 Y-.896 E511.6856*88
> Recv: ok 4040
> Send: N4041 G2 X-5.285 Y-4.264 I-4.417 J1.182 E511.87835*120
> Recv: ok 4041
> Send: N4042 G2 X-2.389 Y-4.572 I-3.352 J-45.435 E511.96033*106
> Recv: ok 4042
> Send: N4043 G2 X1.141 Y-6.394 I-1.539 J-7.312 E512.10744*120
> Recv: ok 4043
> Send: N4044 G1 X1.77 Y-6.604 E512.13167*74
> Recv: ok 4044
> Send: N4045 G3 X14.015 Y-27.814 I4260.911 J2445.778 E513.02769*121
> Recv: ok 4045
> Send: N4046 G3 X14.361 Y-27.888 I0.204 J0.109 E513.04232*122
> Recv: ok 4046
> Send: N4047 G3 X20.135 Y-24.538 I-417298.869 J719165.402 E513.28660*89
> Recv: ok 4047
> Send: N4048 G3 X20.169 Y-24.295 I-0.126 J0.140 E513.29597*88
> Recv: Error:A hit floor
> Changing monitoring state from "Printing" to "Error"
> Send: M112
> Send: N4049 M112*24
> Send: N4050 M104 T0 S0*16
> Send: N4051 M140 S0*85
> Changing monitoring state from "Error" to "Offline after error"
> Connection closed, closing down monitor
> 

[image]

PieBru, 2021-05-23:
Same "catastrophic" behavior here while using Arc Welder as a Cura 4.9.1 plugin or as an OctoPrint plugin, I keep all my software and plugins constantly updated. All runs under Arch Linux.

The same G-Code without the Arc Welder treatment completes regularly, same behaviour with 5 different objects until now.

My printer is a stock hardware TwoTrees Sapphire Plus v1, firmware upgraded with Marlin v2.0.7.2 pre-built by AuroraTech (configs: https://auroratechchannel.com/download/ep10.zip , explanatory video: https://www.youtube.com/watch?v=GbHsNCDrTpY).
When I will have some spare time I will build Marlin v2.0.8.1+ by myself with some additional features I need.

I suspect that the same arcwelded-gcode may give the problem almost randomly at different printing points, but I don't have a sacrificial printer, so I prefer to keep Arc Welder disabled to keep my puppy-printer safe :-) while hoping the RX buffer is enough to not underrun on long arcs.

Nathananan, 2021-06-02:
Just had same issue, half way through print and arm flew off the board. Any updates on this issue? 

Also this issues aside, smashing plugin, usually good results. Great work very much appreciated!

alicephilippa, 2021-06-02:
Usually producing good results. However, like others I find the effector flying off the print and colliding with the edge of the bed. 

Printer: Flsun Q5
Firmware: Marlin 2.0.8.2