FormerLurker/ArcWelderPlugin #152: Slow down to Stutter on small Arcs

Daddycool2708, 2021-02-09:
Everything is working fine. And the prints are much smoother. I noticed a slow down or sometimes a stutter when I have small arc's. Is this normal or is there a way to improve it. 
Anycubic Mega i3 with Knutwurst 1.1.9 (Marlin 2.0.X)

FormerLurker, 2021-02-09:
Hmm.. You could turn min_arc_segments down a bit.  not sure what you have it set to, but 14 is the minimum value that will provide decent results for these arcs.  The default is 24, so there is some room to play around there.  let me know how it goes!

Daddycool2708, 2021-02-10:
Ok I tried several settings but nothing seems to help. It's a Marlin 2.0.5 maybe that causes the issue. But now I will start to look in the GCODE maybe I find something there. I will keep you informed as soon as I found something.

FormerLurker, 2021-02-10:
So, if you can upgrade to at least Marlin 2.0.6  you will have better luck.  There were some issues with acceleration for arcs in prior versions.  I was told by @Thinkyhead that 2.0.6 should be the minimum recommended version.   I do know ppl who had luck with previous versions, but I think it depends on the other firmware settings.

Daddycool2708, 2021-02-10:
it depends on the stl file too. the file which i'm using now in not accurate rounded so on the outer line there is the stuttering. but i think you are right with the frimware version. i will check that too.