FormerLurker/ArcWelderPlugin #153: Feature Request: Add support for TZB fork of Prusa firmware

WarrenSchultz, 2021-02-10:
Here's the string as requested:
```
Send: M115
Recv: FIRMWARE_NAME:Prusa-Firmware TZB3.2.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:BondTech-PE-TZB MK3s EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
```
[TZB Firmware link:](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware)
Thanks!

Naugrimohtar, 2021-06-21:
I second this request. I use this version of Prusa Firmware. Most of the changes are to support a customer MMU2 interface, Would be great if ArcWelder recognized this version of the Prusa firmware.

seantapscott, 2021-06-25:
Arcwelder modifies slicer gcode to compress a series of G1 commands into
arc commands. For a firmware to support arc commands, you need the firmware
to change.

For me, I compiled the firmware from one of the latest Marlin bugfix
branches to increase support of arc commands on an Ender 5.  It did not
require any changes to ArcWelder nor could Arcwelder implement the change
to support your firmware even if they wanted to.

FormerLurker, 2021-06-25:
I think they are talking about the firmware detection.  It is just there for informational purposes, but nobody likes seeing warnings about unknown firmware.  I am almost done with cura arachne support, so I can get to these requests soon.

Also @seantapscott, you are awesome.  I so so appreciate you helping me with these support requests.  You deserve a 🏅

seantapscott, 2021-06-25:
I'm trying to help and provide info where I can.  I noticed you were
responding less, and I hope you're doing great. :)

FormerLurker, 2021-06-25:
I am doing well.  I have been much less responsive lately, due to some craziness with work, family, and some personal stuff, but that will change soon.  End of summer through fall is when I get most of my side programming done, though usually early summer is less busy than this year.  Covid reopenings have kicked things into overdrive, but that will be ending soon.

Normally I don't put things like this into an issue, but it should help to explain the delays.

Naugrimohtar, 2021-06-26:
FormerLurker is right! Just to recognize the software ;) it’s based off of Prusa and arc support is the same as the Prusa build it was based off of