FormerLurker/ArcWelderPlugin #155: Mysterious seams

HansM64, 2021-02-10:
I had a print today that came out with two mysterious seams. I've attached a picture with arrows that point to where the seams showed up. If I print the original gcode the seams are not there.
[image]

FormerLurker, 2021-02-10:
Please post the gcode (both before and after), your firmware version (try sending M115 and pasting in the results), the arc welder plugin version (you can find this in the ArcWelder tab at the bottom), and if you have one, a nice close up shot of your print showing the defects as well as the print without the defects.  That will help immensely.

HansM64, 2021-02-10:
I'm sorry but I cant hand out any files since this is a paid project. I don't think my client would appreciate that. I'll see if I can take a few pictures of the artifacts though.

FormerLurker, 2021-02-10:
How about a single layer where the issue occurs?  One couldn't do much with that.  Without the gcode I'm pretty much unable to do anything unfortunately :(

HansM64, 2021-02-10:
Send: M115
Recv: FIRMWARE_NAME:Marlin Creality 3D SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR-10 Mini EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

;LAYER:100
M117 INDICATOR-Layer100
;TYPE:WALL-INNER
;MESH:LeicaM3grip-18_v6.stl
G2 X194.368 Y116.388 I-9.938 J-14.034 E97.43540 F1513
G1 X194.471 Y116.33 E97.43853
G3 X194.266 Y116.929 I-1.862 J-0.300 E97.44714
G1 X193.911 Y117.829 E97.4729
G2 X192.986 Y117.477 I-837.337 J2202.882 E97.49387
G0 F9000 X188.004 Y119.31
G0 X186.237 Y119.726
G0 X184.779 Y119.896
G0 X174.525 Y120.523
G1 F1513.8 X183.54 Y120.523 E97.73374
G2 X192.954 Y117.501 I-0.017 J-16.229 E98.00107 F1513
G2 X190.843 Y124.106 I44.884 J17.976 E98.18382
G2 X190.570 Y125.665 I23.732 J4.959 E98.22594
G3 X174.525 Y124.051 I-7.944 J-1.586 E98.86124
G1 X174.525 Y120.523 E98.95511
G0 F9000 X173.125 Y120.923
G1 F1513.8 X173.126 Y120.534 E98.96546
G3 X173.221 Y120.278 I3.617 J1.202 E98.97270 F1513
G1 X173.473 Y120.131 E98.98048
G1 X173.556 Y120.123 E98.9827
G1 X183.534 Y120.123 E99.24819
G2 X194.088 Y116.087 I-0.015 J-15.862 E99.55559
G1 X194.381 Y115.92 E99.56456
G3 X194.883 Y116.427 I0.083 J0.419 E99.58680
G3 X194.140 Y118.344 I-21.979 J-7.400 E99.64153
G1 X193.387 Y118.058 E99.66301
G2 X193.199 Y117.986 I-0.171 J0.170 E99.66833
G2 X191.164 Y124.540 I35.628 J14.656 E99.84926
G3 X190.397 Y127.524 I-10.852 J-1.199 E99.93153
G3 X174.125 Y124.066 I-7.772 J-3.443 E100.54801
G1 X174.125 Y120.923 E100.63164
G1 X173.132 Y120.923 E100.65807
G1 X173.125 Y120.923 E100.65825
G0 F9000 X172.725 Y121.323
;TYPE:WALL-OUTER
G1 F1513.8 X172.726 Y120.493 E100.68034
G3 X173.536 Y119.722 I0.807 J0.038 E100.71298 F1513
G1 X183.528 Y119.723 E100.97886
G2 X194.013 Y115.637 I-0.032 J-15.580 E101.28516
G3 X195.179 Y116.796 I0.448 J0.714 E101.34503
G2 X194.371 Y118.859 I121.256 J48.658 E101.40401
G1 X193.434 Y118.504 E101.43068
G2 X191.522 Y124.807 I32.293 J13.232 E101.60617
G3 X186.227 Y132.219 I-8.946 J-0.793 E101.86023
G3 X173.724 Y124.081 I-3.599 J-8.141 E102.33076
G1 X173.725 Y121.323 E102.40415
G1 X172.725 Y121.323 E102.43076
G0 F9000 X172.995 Y121.053
G0 X173.953 Y121.095
G0 X174.605 Y121.447
;TYPE:FILL
G1 F3027.6 X177.758 Y120.602 E102.51762
G0 F9000 X176.761 Y120.602
G1 F3027.6 X186.974 Y130.816 E102.90195
G0 F9000 X187.556 Y130.399
G1 F3027.6 X181.514 Y132.018 E103.06838
G0 F9000 X181.95 Y132.068
G1 F3027.6 X185.043 Y120.524 E103.38638
G0 F9000 X185.159 Y120.515
G1 F3027.6 X190.454 Y125.81 E103.58563
G0 F9000 X190.949 Y123.279
G1 F3027.6 X175.352 Y127.458 E104.01528
G0 F9000 X175.58 Y127.907
G1 F3027.6 X178.792 Y131.119 E104.13614
G0 F9000 X176.495 Y129.244
G1 F3027.6 X178.81 Y120.602 E104.3742
G0 F9000 X186.237 Y119.726
G0 X191.595 Y118.466
G1 F3027.6 X192.26 Y119.131 E104.39922
G0 F9000 X191.845 Y118.319
G1 F3027.6 X188.998 Y128.945 E104.69193
G0 F9000 X192.451 Y118.346
G0 X192.362 Y118.257
;TYPE:SKIN
G1 F1513.8 X192.487 Y118.099 E104.69676
G0 F9000 X192.527 Y118.139
G0 X193.445 Y117.36
G0 X193.529 Y117.444
G1 F1513.8 X193.729 Y117.362 E104.70162
G1 X193.967 Y117.034 E104.71075
G0 F9000 X193.919 Y116.985
G0 X194.206 Y115.882
G0 X194.191 Y115.843
G1 F1513.8 X194.126 Y115.779 E104.71196
G0 F9000 X194.172 Y115.895
G0 X192.108 Y117.405
G0 X190.537 Y118.309
G0 X188.863 Y119.026
G0 X187.127 Y119.544
G0 X185.337 Y119.858
G0 X174.364 Y120.341
G0 X174.342 Y120.319
G1 F1513.8 X174.201 Y120.46 E104.71515
G1 X173.49 Y120.598 E104.72674
G1 X173.328 Y120.719 E104.72998
G1 X172.97 Y120.361 E104.73808
G1 X173.218 Y120.043 E104.74453
G0 F9000 X173.474 Y120.346
;MESH:NONMESH
G0 F300 X173.474 Y120.346 Z16.32
G0 F9000 X184.435 Y119.936
G0 X186.237 Y119.726
G0 X188.004 Y119.31
G0 X192.986 Y117.478
;TIME_ELAPSED:6386.072673

HansM64, 2021-02-10:
Send: M115
Recv: FIRMWARE_NAME:Marlin Creality 3D SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR-10 Mini EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

;LAYER:100
M117 INDICATOR-Layer100
;TYPE:WALL-INNER
;MESH:
G2 X194.368 Y116.388 I-9.938 J-14.034 E97.43540 F1513
G1 X194.471 Y116.33 E97.43853
G3 X194.266 Y116.929 I-1.862 J-0.300 E97.44714
G1 X193.911 Y117.829 E97.4729
G2 X192.986 Y117.477 I-837.337 J2202.882 E97.49387
G0 F9000 X188.004 Y119.31
G0 X186.237 Y119.726
G0 X184.779 Y119.896
G0 X174.525 Y120.523
G1 F1513.8 X183.54 Y120.523 E97.73374
G2 X192.954 Y117.501 I-0.017 J-16.229 E98.00107 F1513
G2 X190.843 Y124.106 I44.884 J17.976 E98.18382
G2 X190.570 Y125.665 I23.732 J4.959 E98.22594
G3 X174.525 Y124.051 I-7.944 J-1.586 E98.86124
G1 X174.525 Y120.523 E98.95511
G0 F9000 X173.125 Y120.923
G1 F1513.8 X173.126 Y120.534 E98.96546
G3 X173.221 Y120.278 I3.617 J1.202 E98.97270 F1513
G1 X173.473 Y120.131 E98.98048
G1 X173.556 Y120.123 E98.9827
G1 X183.534 Y120.123 E99.24819
G2 X194.088 Y116.087 I-0.015 J-15.862 E99.55559
G1 X194.381 Y115.92 E99.56456
G3 X194.883 Y116.427 I0.083 J0.419 E99.58680
G3 X194.140 Y118.344 I-21.979 J-7.400 E99.64153
G1 X193.387 Y118.058 E99.66301
G2 X193.199 Y117.986 I-0.171 J0.170 E99.66833
G2 X191.164 Y124.540 I35.628 J14.656 E99.84926
G3 X190.397 Y127.524 I-10.852 J-1.199 E99.93153
G3 X174.125 Y124.066 I-7.772 J-3.443 E100.54801
G1 X174.125 Y120.923 E100.63164
G1 X173.132 Y120.923 E100.65807
G1 X173.125 Y120.923 E100.65825
G0 F9000 X172.725 Y121.323
;TYPE:WALL-OUTER
G1 F1513.8 X172.726 Y120.493 E100.68034
G3 X173.536 Y119.722 I0.807 J0.038 E100.71298 F1513
G1 X183.528 Y119.723 E100.97886
G2 X194.013 Y115.637 I-0.032 J-15.580 E101.28516
G3 X195.179 Y116.796 I0.448 J0.714 E101.34503
G2 X194.371 Y118.859 I121.256 J48.658 E101.40401
G1 X193.434 Y118.504 E101.43068
G2 X191.522 Y124.807 I32.293 J13.232 E101.60617
G3 X186.227 Y132.219 I-8.946 J-0.793 E101.86023
G3 X173.724 Y124.081 I-3.599 J-8.141 E102.33076
G1 X173.725 Y121.323 E102.40415
G1 X172.725 Y121.323 E102.43076
G0 F9000 X172.995 Y121.053
G0 X173.953 Y121.095
G0 X174.605 Y121.447
;TYPE:FILL
G1 F3027.6 X177.758 Y120.602 E102.51762
G0 F9000 X176.761 Y120.602
G1 F3027.6 X186.974 Y130.816 E102.90195
G0 F9000 X187.556 Y130.399
G1 F3027.6 X181.514 Y132.018 E103.06838
G0 F9000 X181.95 Y132.068
G1 F3027.6 X185.043 Y120.524 E103.38638
G0 F9000 X185.159 Y120.515
G1 F3027.6 X190.454 Y125.81 E103.58563
G0 F9000 X190.949 Y123.279
G1 F3027.6 X175.352 Y127.458 E104.01528
G0 F9000 X175.58 Y127.907
G1 F3027.6 X178.792 Y131.119 E104.13614
G0 F9000 X176.495 Y129.244
G1 F3027.6 X178.81 Y120.602 E104.3742
G0 F9000 X186.237 Y119.726
G0 X191.595 Y118.466
G1 F3027.6 X192.26 Y119.131 E104.39922
G0 F9000 X191.845 Y118.319
G1 F3027.6 X188.998 Y128.945 E104.69193
G0 F9000 X192.451 Y118.346
G0 X192.362 Y118.257
;TYPE:SKIN
G1 F1513.8 X192.487 Y118.099 E104.69676
G0 F9000 X192.527 Y118.139
G0 X193.445 Y117.36
G0 X193.529 Y117.444
G1 F1513.8 X193.729 Y117.362 E104.70162
G1 X193.967 Y117.034 E104.71075
G0 F9000 X193.919 Y116.985
G0 X194.206 Y115.882
G0 X194.191 Y115.843
G1 F1513.8 X194.126 Y115.779 E104.71196
G0 F9000 X194.172 Y115.895
G0 X192.108 Y117.405
G0 X190.537 Y118.309
G0 X188.863 Y119.026
G0 X187.127 Y119.544
G0 X185.337 Y119.858
G0 X174.364 Y120.341
G0 X174.342 Y120.319
G1 F1513.8 X174.201 Y120.46 E104.71515
G1 X173.49 Y120.598 E104.72674
G1 X173.328 Y120.719 E104.72998
G1 X172.97 Y120.361 E104.73808
G1 X173.218 Y120.043 E104.74453
G0 F9000 X173.474 Y120.346
;MESH:NONMESH
G0 F300 X173.474 Y120.346 Z16.32
G0 F9000 X184.435 Y119.936
G0 X186.237 Y119.726
G0 X188.004 Y119.31
G0 X192.986 Y117.478
;TIME_ELAPSED:6386.072673

FormerLurker, 2021-02-10:
OK, that is helpful.  Can you also post the non welded version of layer 100?  

Also, your firmware doesn't report the version unfortunately.   I'd recommend flashing the latest stable release of marlin if you can, even if that isn't the problem here.  You'll get better results.  The version you have may be a fork of Marlin 1, but 2.0.6 or higher is recommended for printing arcs according to the creator of Marlin.  There have been other fixes in later releases too.

FormerLurker, 2021-02-10:
OH, also, if you can find the coordinate of the printhead right before layer 100, that would also help.  The first command on that layer is a G2, and without the starting point it's difficult to see what's going on with it.  I basically need just the X and Y coordinate from the nearest G0/G1/G2/G3 right above the ```;LAYER:100``` comment.  It might not matter, but it's usually helpful to look at the entire layer.

HansM64, 2021-02-10:
Here's 10 layers from both files. I noticed that it's not doing it on every layer. Also I attached some pictures. Sorry about the quality. It's was hard to ger the camera on my mobile to focus right. The last picture is the original code.

;LAYER:101
M117 INDICATOR-Layer101
;TYPE:WALL-INNER
;MESH:...
G2 X194.411 Y116.353 I-13.544 J-18.637 E104.79290 F1504
G1 X194.467 Y116.333 E104.79449
G1 X194.462 Y116.376
G3 X193.911 Y117.828 I-8.838 J-2.520 E104.82944
G3 X192.985 Y117.477 I209.597 J-553.752 E104.85041
G0 F9000 X188.006 Y119.31
G0 X186.238 Y119.726
G0 X184.777 Y119.896
G0 X174.525 Y120.523
G1 F1504.8 X183.54 Y120.523 E105.09029
G2 X192.955 Y117.5 I0.008 J-16.147 E105.35764 F1504
G3 X192.764 Y118.009 I-36.161 J-13.208 E105.36971
G2 X190.735 Y124.673 I35.304 J14.395 E105.55533
G3 X187.911 Y130.219 I-8.260 J-0.713 E105.72501
G3 X174.525 Y124.051 I-5.283 J-6.142 E106.21741
G1 X174.525 Y120.523 E106.31129
G0 F9000 X173.125 Y120.923
G3 X173.222 Y120.277 I0.909 J-0.193 E106.32895 F1504
G1 X173.476 Y120.13 E106.33676
G1 X173.55 Y120.123 E106.33874
G1 X183.534 Y120.123 E106.60439
G2 X194.169 Y116.016 I-0.010 J-15.852 E106.91463
G2 X194.457 Y115.909 I-3.364 J-9.530 E106.92284
G3 X194.879 Y116.251 I0.005 J0.424 E106.93844
G1 X194.842 Y116.561 E106.94673
G2 X194.140 Y118.344 I93.448 J37.748 E106.99773
G1 X193.387 Y118.058 E107.0192
G2 X193.199 Y117.986 I-0.171 J0.170 E107.02451
G2 X191.128 Y124.740 I35.051 J14.442 E107.21090
G3 X187.688 Y130.908 I-8.630 J-0.769 E107.40434
G3 X174.125 Y124.066 I-5.062 J-6.830 E107.90420
G1 X174.125 Y120.923 E107.98784
G1 X173.13 Y120.923 E108.01431
G1 X173.125 Y120.923 E108.01444
G0 F9000 X172.725 Y121.323
;TYPE:WALL-OUTER
G1 F1504.8 X172.726 Y120.493 E108.03653
G3 X173.533 Y119.722 I0.807 J0.037 E108.06913 F1504
G1 X183.528 Y119.723 E108.33506
G2 X193.933 Y115.689 I-0.003 J-15.449 E108.63885
G3 X195.221 Y116.689 I0.527 J0.649 E108.69822
G2 X194.371 Y118.859 I126.513 J50.807 E108.76023
G1 X193.434 Y118.504 E108.78691
G2 X191.356 Y125.816 I30.774 J12.692 E108.98962
G3 X173.724 Y124.081 I-8.730 J-1.736 E109.68699
G1 X173.725 Y121.323 E109.76039
G1 X172.725 Y121.323 E109.787
G0 F9000 X172.995 Y121.053
G0 X173.953 Y121.095
G0 X174.605 Y121.563
;TYPE:FILL
G1 F3009.6 X178.194 Y120.602 E109.88586
G0 F9000 X178.693 Y120.601
G1 F3009.6 X176.406 Y129.137 E110.12099
G0 F9000 X175.407 Y127.574
G1 F3009.6 X179.124 Y131.291 E110.26086
G0 F9000 X181.802 Y132.058
G1 F3009.6 X187.318 Y130.58 E110.41281
G0 F9000 X187.069 Y130.751
G1 F3009.6 X176.92 Y120.602 E110.79471
G0 F9000 X175.401 Y127.562
G1 F3009.6 X190.919 Y123.404 E111.22218
G0 F9000 X190.485 Y125.681
G1 F3009.6 X185.306 Y120.503 E111.41705
[... 4280 lines cut ...]
G0 X174.201 Y120.46
G1 F1475.7 X174.342 Y120.319 E177.9612
G0 F9000 X174.32 Y120.297
G0 X184.455 Y119.935
G0 X186.264 Y119.722
G0 X186.766 Y119.603
G0 X192.272 Y118.167
G0 X192.361 Y118.257
G1 F1475.7 X192.487 Y118.1 E177.9661
G0 F9000 X192.446 Y118.059
G0 X193.445 Y117.36
G0 X193.528 Y117.444
G1 F1475.7 X193.729 Y117.362 E177.96807
G1 X193.968 Y117.035 E177.97173
G1 X194.673 Y116.326 E177.98077
G1 X195.026 Y116.113 E177.98449
G0 F9000 X194.665 Y116.243
;MESH:NONMESH
G0 F300 X194.665 Y116.243 Z17.92
G0 F9000 X192.986 Y117.478
;TIME_ELAPSED:6517.318808

[image]
[image]
[image]

FormerLurker, 2021-02-11:
OK, that's probably enough for me to go on.  I may try to print those layers to see what happens with my firmware in addition to running this through the debugger.